modelica/ModelicaStandardLibrary #3739: Improve inheritance of classes in MultiBody

tobolar, 2021-01-29:
The inheritance of classes is not done consequently.
See for example #3737 or https://github.com/modelica/ModelicaStandardLibrary/issues/2171#issuecomment-280661422

beutlich, 2021-01-29:
> 
> 
> The inheritance of classes is not done consequently.
> See for example #3737 or [#2171 (comment)](https://github.com/modelica/ModelicaStandardLibrary/issues/2171#issuecomment-280661422)

Or also https://github.com/modelica/ModelicaStandardLibrary/pull/3725#issuecomment-762969249

tobolar, 2021-02-01:
> Or also https://github.com/modelica/ModelicaStandardLibrary/pull/3725#issuecomment-762969249

I would proceed with Constraints once #3725 is merged.

tobolar, 2021-03-12:
I have started to apply the inheritance properly but see currently two problems.

1. I really stumble on the usage of `cardinality` in the MultiBody classes. It is generally used in base classes such as `Interfaces.PartialOneFrame_b` but not in the classes which could inherit it, e.g. `Parts.Fixed`. Moreover, the `cardinality` is deprecated and even if #1410 gets some solution, it is obviously yet not resolved. So we can
   1. Wait untill #1410 is solved ot
   2. Implement a "half" solution for classes only where `cardinality` is used already.

2. The connector's description in the particular classes is moslty more specific compared to a base class. E.g. `frame_b` in `Parts.Fixed` is sayed to be "Coordinate system fixed in the world frame", but it is "Coordinate system fixed to the component with one cut-force and cut-torque" in `Interfaces.PartialOneFrame_b`.
   Any possibility to retain the more precise text?

beutlich, 2021-03-12:
It should (?) be egal Modelica to overwrite the description in the extends statement, see for example
```mo
model M "Extend from PartialTwoFrames, but set other frame descriptions"
  extends Modelica.Mechanics.MultiBody.Interfaces.PartialTwoFrames(
    frame_a "This is frame A ...",
    frame_b "... and this is frame B");
end M;
```

christiankral, 2021-03-15:
This is somehow related with #2455

tobolar, 2021-03-16:
> It should (?) be egal Modelica to overwrite the description in the extends statement...

@beutlich Thanks, I will use this.

TManikantan, 2023-03-09:
@tobolar Hi!!! ,I recently assumed the responsibility of project deputy. Did you happen to fix the issue? would you please commit the fix,. We can get it reviewed and merge it.

tobolar, 2023-03-09:
I started this a long time ago but let it be due to lack of time. Will try to recall.