modelica/ModelicaStandardLibrary #3746: A lot of type conversion warnings when compiling ModelicMatIO.c with Visual Studio on x64

gkurzbach, 2021-02-11:
When compiling ModelicaMat.c with Visual Studio on x64 with warn level 3 in debug mode there are a lot of warnings like:
```
1>D:\ITI-Projekte\SimulationX_WT\Application\src\Dlls\ModelicaMatIO\ModelicaMatIO.c(2929,13): warning C4267: "=": Konvertierung von "size_t" nach "int", Datenverlust möglich
1>D:\ITI-Projekte\SimulationX_WT\Application\src\Dlls\ModelicaMatIO\ModelicaMatIO.c(2929,13): warning C4267: "*=": Konvertierung von "size_t" nach "int", Datenverlust möglich
1>D:\ITI-Projekte\SimulationX_WT\Application\src\Dlls\ModelicaMatIO\ModelicaMatIO.c(2929,13): warning C4267: "Funktion": Konvertierung von "size_t" nach "long", Datenverlust möglich
1>D:\ITI-Projekte\SimulationX_WT\Application\src\Dlls\ModelicaMatIO\ModelicaMatIO.c(2929,13): warning C4267: "+=": Konvertierung von "size_t" nach "int", Datenverlust möglich
1>D:\ITI-Projekte\SimulationX_WT\Application\src\Dlls\ModelicaMatIO\ModelicaMatIO.c(2935,13): warning C4267: "=": Konvertierung von "size_t" nach "int", Datenverlust möglich
...
```

When inspecting the code one can see the compiler is right. At many places there is a mixture of int and size_t. This does not show a good code quality.
(surprisingly when compiling in release mode there are no such warnings)

beutlich, 2021-02-11:
You can disable thess kind of conversion warnings, which is what previously was done in SimulationX. Anyway, the issue is an upstream issue for https://matio.sourceforge.io/, but it is tedious work to do it right and not just to add the type casts.

gkurzbach, 2021-02-11:
Yes, I dit it (again). If this issue is moved to  https://matio.sourceforge.io/  you can close this ticket.

beutlich, 2024-01-16:
Resolved upstream by https://github.com/tbeu/matio/commit/634dd20640ba53daf91c7e4ba5c5a064f32450a5.

closed 2024-01-16 by beutlich