modelica/ModelicaStandardLibrary #3747: Node temperatures of pipe not sufficient for BasicHX

PerNil, 2021-02-12:
The BasicHX example becomes quite inaccurate at high differences in inlet and outlet temperature. The reason seems to be that the node temperatures in the fluid are staggered in the pipes of opposing directions, meaning that the positions along the pipes do not correspond over the wall. For this purpose, it could have been better if the states along the pipe represented averages over the volumes. I have been looking for a simple workaround by calculating the average temperatures, but that would require using the inlet temperature inside PartialPipeFlowHeatTransfer.

justnielsen, 2021-02-12:
@PerNil Using the average volume as a dynamic state has an unwated consequence, seen from a dynamic perspective: it introduces  "right hand side zeros" (non-minimum phase). In practice this means that if you abruptly increase the inlet temperature of a cell, the outlet temperature will take a 'dip' before it rises to its steady-state value since the state (inlet/outlet average temperature) will react slowly.

PerNil, 2021-02-12:
OK. I am not proposing to change the states used in the pipe, but maybe there could be an alternative to using the staggered temperatures for the heat flux calculation?

casella, 2021-03-25:
The temperature state of each volume is better taken as the outlet temperature, to avoid the effect @justnielsen mentioned. However, one can use the average temperature (T_in + T_out)/2 of each volume to compute the heat transfer. This provides better approximation of the heat transfer, provided that the NTU of each volume is less than one, otherwise you may get unphysical results, including violations of the second principle of thermodynamics.

The ThermoPower library implements all these options, including one that continuously shifts from average temperature to outlet temperature for heat transfer computation when the flow rate gets below a certain limit, to avoid the latter effect.

Unfortunately I have no time to try implementing this in Modelica.Fluid anytime soon. If someone else does, I would be happy to help and give advice.

PerNil, 2021-03-26:
Thanks!

I do not have the expertise to implement it, especially not since I think a general solution would touch Modelica.Fluid.Pipes.BaseClasses.

In the mean time my workaround might be useful for someone.
I changed the Q_flow calculation in a copy of PartialPipeFlowHeatTransfer to this:
  input Modelica.Units.SI.Temperature Tinca "Inlet temperature from state_a.T";
  Q_flows[1:1]={alphas[1]*surfaceAreas[1]*(heatPorts[1].T - (Tinca+Ts[1])/2)*nParallel};
  Q_flows[2:n]={alphas[i]*surfaceAreas[i]*(heatPorts[i].T - (Ts[i-1]+Ts[i])/2)*nParallel for i in 2:n};
Then I made a copy of LocalPipeFlowHeatTransfer that extends it, which can be called in the pipe models.
It is not bounded as mentioned above but seems to work in controlled conditions.