modelica/ModelicaStandardLibrary #3748: Inconsistent M_OlIne in MSL (also 4.0)

max-privato, 2021-02-12:
Consider first Oline. It contains what is usually called "cascading-pi" model. 
The model takes as reference for voltage pin p3 => this is ok for me.

Consider now M_Oline.  Its impedance and susceptance matrices are written with reference to the internal ground M, which is not sent to the outside of the model.

In fact, the picture shown in the documentation indicates that the matrices are written to reference to ground.  In reality, they could be written with reference to any terminal, not only the circuit ground or not. 

Consider that there is a conceptual difference between the circuit ground and the physical earth below our feet. Let's call it earth and keep distinct from ground). While line parameters are usually written assuming as reference the earth (taken as a connector), the circuital ground (reference for voltages) may be well chosen differently. I mean, there is no need to oblige the modeler to write the line parameters with reference to an electric point which is exactly the circuit ground.

Much better would be to follow for M_Oline the same approach used for Oline, and using a "neutral" pin, to be put in the lower side of the box of the model, so that it can be connected to any conductor, possibly not the circuit ground.
In practice, the internal pin presently named M should be actually a pin, instead of a ground, and connected to the outside, pretty much the same as already done with OLine.

BTW, in the picture shown in M_Oline, according to circuit drawing standards, C32 is not connected to the negative pin of L3, but only with the negative pin of C43. We should add a dot (a small filled circle) to indicate the connection. The same with similar connections in the same picture.

dietmarw, 2021-02-18:
I agree that not having the ground hard-coded inside the `M_OLine` model but rather having it as a port that then can be connected outside would give the model more flexibility and also make it more consistent with the `OLine` model. I fear this would be a non-backward compatible change and it probably would not be possible to not break user models using this changed components since then a new additional ground would have to be introduced in user models to keep the (then) old behaviour. 

I see the following options:
 1. Change the existing `M_OLine` and add a warning in the release notes about this non-backward compatible change
 2. Introduce a new variant (e.g.,  `M_OLineNew`) and add a deprecation warning to `M_OLine` which then will be removed/replaced by the new variant in a new major version of the MSL.
 
Both option should be feasible for the next minor version. Yes even 1. since there is no option that could be fixed with a conversion script. So no need to wait for a major version if we decide to break user models in this case.

As for the diagram, it probably should be redone completely to use the dot for connection convention rather than the cross over = no connect and touch = connect.

HansOlsson, 2021-02-18:
Would it be possible to have a conditional connector for grounding similarly as in the Rotational-library?

dietmarw, 2021-02-18:
Actually, that would be a solution allowing backward compatibility. Haven't thought about that but you are right @HansOlsson

max-privato, 2021-02-18:
Glad that my ticket was taken seriously into consideration.

But why the 5.0.0. milestone? In case a conditional connector is used there would be no backward compatibility issue and therefore the modification can be scheduled for 4.1, which I presume is programmed? The documentation diagram could be fixed in 4.1 as well.

In that case, in 5.0.0. maybe the internal ground could be totally removed. I propose this since I think it is not a good idea to hide a ground inside a model. If this is not deemed acceptable, at least the default could be changed from internal ground into a connector.

dietmarw, 2021-02-19:
Yes you are right it could be done for the next minor release since no conversion would be needed or even possible. Which leads to that it is actually not possible to remove the internal ground (ever) without breaking user models. As far as I can tell, conversion script can not add a new component to a model and that would be needed if you remove the internal ground. in order to place it on the outside. So that just leaves to break backward compatibility on purpose and I don't think of having the internal ground is reason enough to do that. But that's for the library officers to decide at one point.

dietmarw, 2021-02-19:
@max-privato Feel free to submit a PR with the conditional connector.

max-privato, 2021-02-20:
> Yes you are right it could be done for the next minor release since no conversion would be needed or even possible. Which leads to that it is actually not possible to remove the internal ground [...]

Maybe I'm wrong, but I don't see too difficult for a conversion script to substitute the row 
    Modelica.Electrical.Analog.Basic.Ground M;
with
  Modelica.Electrical.Analog.Interfaces.Pin p3
(if we want the same name as OLine to be used).

We''ll see when we are near enough to 5.0 release date. 
At that time it may also be possible. if there are too big difficulties in doing this change through a conversion script, that several other models need to be updated for which this cannot be done converted through scripts. This one, in that case, would be just one more and may be included.

max-privato, 2021-02-20:
> @max-privato Feel free to submit a PR with the conditional connector.
Ok. I'll do this

max-privato, 2021-02-20:
> Yes you are right it could be done for the next minor release since no conversion would be needed or even possible. Which leads to that it is actually not possible to remove the internal ground [...]
ok.

However, in principle, when we are near enough to 5.0 release date it may be possible that there are other models needing to be updated in a way that cannot be done through scripts. M_OLine, in that case, would just be one more and may be included. Obviously, it is just a theoretical hypothesis.

AHaumer, 2021-02-21:
Actually I strongly prefer the solution with the optional intrnal ground and optional ground connector. 
I'd include it in #3755 otherwise the PRs depend on each other.

max-privato, 2021-02-22:
> Actually I strongly prefer the solution with the optional intrnal ground and optional ground connector.
> I'd include it in #3755 otherwise the PRs depend on each other.

I don't understand well: do you prefer making the PR on this ticket, or that I do this?
Just to be sure of not doing useless work.

My idea was to give users choice (in the end I agree with this), and try to make the internal ground visible also from the M_OLine icon when it is active.
So when one chooses internal ground they should see a ground sign on the icon, otherwise a connector for the reference pin (that can optionally be connected to an outside ground)

Never done conditional icons, but I could try.

dietmarw, 2021-02-22:
@max-privato  No action needed by you any more since @AHaumer beat you to it by creating the PR #3755 already.

AHaumer, 2021-02-22:
Well I've choosen to avoid showing the conditional ground in the icon.
Either you see the connector -> internal ground conditionally removed,
or you don't see the connector -> internal ground conditionally instantiated.

max-privato, 2021-02-22:
> Well I've choosen to avoid showing the conditional ground in the icon.
> Either you see the connector -> internal ground conditionally removed,
> or you don't see the connector -> internal ground conditionally instantiated.
Nice that you've done the work I intended to do.
For me  your choice is ok, even though, as you saw, I had a different program.

I hope you've fixed also the diagrams in the documentation. 
Not only dots to show electrical connecitons were missing, but also loops to show mutual couplings.

AHaumer, 2021-02-23:
Feel free to contribute to my PR with improving the documentation. 
As far as I see the image is a screenshot from a "dummy" model and I think it's hard work to improve it - maybe necessary?
To be honest, I had the focus on functionality.

PR #3815 (2021-05-26): Fix inconsistent M_OLine model
(reimplementation of #3756, fixes #3748)

Now I've included the desired documentation and images. Feel free to contribute to this PR, I consider this as a **draft**.
We are discussing **peanuts** (whether the ground can be accessed or not) compared to to the **open issues**:
- initialization of inductances and capacitances is not implemented
- a test / demo example is missing

I have neither the time to dig into the initialization nor the specific knowledge to implement a meaningful example.
@dietmarw and @max-privato, I hope you are willing to review and contrib [... cut]
commit 3754abb7 (2021-11-16): Fix Inconsistent M_OLine as reported in #3748

Fixes #3748 for 4.0.1
closed 2021-05-26 by dietmarw