modelica/ModelicaStandardLibrary #3749: What does an empty annotation mean, if anything?

maltelenz, 2021-02-15:
There seem to be interpretations that adding an empty annotation like `annotation(Dialog)` means something different than having no annotation at all, see https://github.com/modelica/ModelicaStandardLibrary/issues/3726#issuecomment-762823785 and the discussion following it.

If this is the intended interpretation, should it be made explicit in the specification:
* What does it mean?
* Is there a difference between `annotation(Dialog)` and `annotation(Dialog())`? Are both OK?
* For which annotations does an empty annotation mean something?

henrikt-ma, 2021-02-15:
I don't believe it's the most central question being asked here, but I'd expect `annotation(Dialog)` to be rejected as malformed, just like this model:
```
model NoRecordArgumentList
  record R
    Real x = 1.0;
  end R;
  R r = R; /* Missing record constructor argument list! */
end NoRecordArgumentList;
```

I hope that this is clear enough that it doesn't even require further clarification.

maltelenz, 2021-02-15:
> I'd expect `annotation(Dialog)` to be rejected as malformed

I would agree.

> I hope that this is clear enough that it doesn't even require further clarification.

It is used in MSL, so it wasn't clear to at least one person/tool:
https://github.com/modelica/ModelicaStandardLibrary/blob/403e7c1e9c9399fe0481e593815deb4bae094203/Modelica/Mechanics/MultiBody/Visualizers/Advanced/Arrow.mo#L21

henrikt-ma, 2021-02-15:
Regarding an empty argument list, I'm afraid this becomes a separate issue for each annotation.

Defining semantics of an annotation often requires a mix of grammar or pseudo-code record, and a textual description of valid use and special defaults that can't be expressed in grammar or pseudo-code.  Note also that it is standard to have semantics depending on the presence of a value.

When designing the new annotations for figures we put much effort into making handling of defaults as explicit as possible, by specifying defaults in the pseudo-code record definitions, rather than relying on textual descriptions of what it means to omit a member in the record constructor.  However, it didn't take me long now to see that we failed already at `Figure.caption`, which doesn't have a default in the pseudo-code record definition, and no textual description of what it means to leave it out. :(  Just like this deserves its own issue being opened, I'm afraid there is no better solution at this point than also opening issues for any other annotation with unclear defaults – a complete redesign of annotation semantics is probably as unlikely to happen as the proposal to formalize annotation types using algebraic data types.

henrikt-ma, 2021-02-15:
Technically, I don't see that it would be necessary for the default behavior to be the same as the behavior given by an annotation with empty argument list.  However, it still seems like a pretty natural assumption to assume that they would be equal unless anything else is stated, and I think it would be great if we could make this a general rule about annotations defined by pseudo-code records.  (Then, it would be good if we made use of pseudo-code records as often as possible, to avoid special case-by-case semantics for annotations with structure described by grammar.)