modelica/ModelicaStandardLibrary #3750: The Dialog.caption should be clearly optional

henrikt-ma, 2021-02-15:
The annotations for figures are systematically trying to make default semantics as clearly defined as possible, for example by providing default values in the pseudo-record definitions as often as possible to make it clear that no value has to be provided.  When no default value is provided, and unless it's an array member of flexible size, the textual description should always describe the semantics of omitting a value when using the annotation.

According to this silently adopted convention, the `Dialog.caption` would be mandatory, but that was not the intention.

I suggest that the empty string is explicitly made the default of `caption`, and I don't see a need to provide any special meaning to the empty string in this case.  Note that this would mostly be a backwards compatible change.

henrikt-ma, 2021-02-15:
Actually, the same applies to `identifier`.  It should have the empty string as default in both `Figure` and `Plot`, but in this case the empty string should be given the special meaning of not having an identifier at all: the figure or plot will not be possible to reference, and the uniqueness requirement doesn't apply.

beutlich, 2021-02-15:
Should this be a specification issue?

closed 2021-02-15 by henrikt-ma