modelica/ModelicaStandardLibrary #3754: Icon design guidelines for connectors will cause "steps"

dietmarw, 2021-02-18:
When looking at Fig. 8 in `Modelica.UsersGuide.Conventions.Icons` I noticed that the positioning of the output connectors are suboptimal for the default grid resolution of 2x2. 
[image]

Having the output connectors 3 grids offset means that when connecting there will always be a step introduced in the connecting line which can not be avoided unless one sets the grid resolution manually to 1x1. This is not ideal and I would suggest to update the guideline (and the component) to:

[image]

christiankral, 2021-02-21:
> Having the output connectors 3 units offset means that when connecting there will always be a step introduced in the connecting line which can not be avoided unless one sets the grid resolution manually to 1x1. 

@dietmarw could you please provide a graphical example of (a) the original model and (b) the modified model, as I am not sure if I 100% understand this issue.

dietmarw, 2021-02-22:
No I can't since I now suddenly can reproduce the problem even though having struggled with it a lot of times. Pleas leaver this open for now and I will either come back with an update or close it as non-issue.

dietmarw, 2022-03-01:
I revisited this and it is actually a non-issue. The default scale for inserting components is 0.1 and the default grid size is 200x200. This means as long as the connectors are positions at multiple of 20 points on the icon grid it will be fine.

closed 2022-03-01 by dietmarw

henrikt-ma, 2022-03-01:
I think I might know what problem you were thinking of.  At least in System Modeler, the default grid spacing in diagram view is 5mm, and diagrams look nice if connection lines are drawn on this grid.  For a component scaled down by the factor 0.1, this means that connectors should be preferred on a grid with spacing 50mm.

This becomes a problem if the suggestion is based on a grid with spacing 20mm.

In System Modeler, we have solved this by having a grid separation of 10mm, with thicker grid lines on the multiples of 50mm.

dietmarw, 2022-03-01:
@henrikt-ma Just out of interest. What makes you speak of resolution in "mm" instead of "pt". Is that specified anywhere? Just seems strange to have some digital format be defined in mm. Normally you have points or pt and only if you want to print it to some physical medium you would scale it with the help of a dpi factor (oh the imperial irony) to mm.

sjoelund, 2022-03-01:
https://specification.modelica.org/master/annotations.html#common-definitions

dietmarw, 2022-03-01:
Ah OK, thanks @sjoelund for pointing it out. Even though I still thinks the unit makes no sense here ;-)

henrikt-ma, 2022-03-01:
For me, working with things like line thickness and font sizes make most sense in a setting with physical dimensions on the canvas.

dietmarw, 2022-03-01:
...and font sizes are normally in pt too. I've never seen font sizes being specified in mm size.

henrikt-ma, 2022-03-01:
But that pt is a physical length, not a pixel.

dietmarw, 2022-03-02:
Ah yes, forgot about that.