modelica/ModelicaStandardLibrary #3759: Modelica Standard Tables in the FMI set/get state context

agnesramle, 2021-02-23:
In the FMI 2.0 specification there is the concept of setting and getting FMU State. This doesn’t fit very well with Modelica Standard Tables since there are parts of the struct CombiTimeTable that are not allocated when calling the constructor (ModelicaStandardTables_CombiTimeTable_init) but in ModelicaStandardTables_CombiTimeTable_nextTimeEvent. To make sure that the constructor is called when setting FMU state on an FMU that is not yet initialized or has been reset is pretty straightforward to do. However, the compiler has no way to know that nextTimeEvent needs to be called in order to allocate memory nor if it needs to be called several times to reach the point where the state is saved. Are there any intentions to update this or are there thoughts/ideas on how to handle it?

beutlich, 2021-02-23:
Isn't this a general issue with FMI and Modelica external objects, and as the Modelica Standard Tables happen to be the only external objects in MSL, it is only observed there? Hence I wonder if it is a duplicate issue of MO#1950 or really specific to the CombiTimeTable implementation.

beutlich, 2021-11-27:
Closing since there was no more feedback from author.

closed 2021-11-27 by beutlich