modelica/ModelicaStandardLibrary #3772: PowerConverters...Delta: Bad icon and name

max-privato, 2021-03-11:
I think that the component
Modelica.MultiPlase.Basic.Multidelta
has a wrong icon and name. What it actually does is a phase rotation, and therefore I would recommend as name PhaseRotation
The icon would be something like the following picture (possibly omitting the numbers)
[image]
I'm aware that changing the name would imply this change, if accepted, to be scheduled for next major revision, for compatibility reasons.

AHaumer, 2021-03-11:
First, MultiPhase has been renamed to Polyphase,
Second, would you please explain how would you implement a delta-connection (threephase) resp. a polygon-connection (polyphase)? Correct, a phase rotation. 
end of phase 1 connected to begin of phase 2
end of phase 2 connected to begin of phase 3
end of phase 3 connected to begin of phase 1
or as alternative:
begin of phase 1 connected to end of phase 2
begin of phase 2 connected to end of phase 3
begin of phase 3 connected to end of phase 1
So yes, you _could_ change the name to PhaseRotation. This would break backwards compatibility. As you said, we'd have to wait for the next major release. But why should we do that? The most important application is to establish a delta-connection (threephase) resp. a polygon-connection (polyphase). So in my humble opinion we could change the name to polygon. 
There are only a few applications for phase rotation (have a look at the transformers ...)
Well, let's keep the old name.

max-privato, 2021-03-15:
I take as example Electrical.PowerConverters.ACDC.Control.VoltageBridge2mPulse.
In its delta plug_p has just pins rotated with respect to plug_n.
Component voltageSensor, which takes voltages between corresponding pins _does not_ create a delta connection: a delta connection would create a short circuit for third harmonic currents, which this use of "delta" does not do at all.
That's why, for me, the name delta is profoundly wrong.
The icon is even wronger since what is inside the component is not at all what the icon suggests. That's why I proposed an alternative icon.