modelica/ModelicaStandardLibrary #3773: spatialDistribution not listed in ModelicaReference

beutlich, 2021-03-12:
ModelicaReference misses the spatialDistribution operator of MLS 3.3.

dietmarw, 2021-03-15:
I was wondering when fixing this if using SVG directly would be allowed already for the next MSL release. This would allow me to include the equations displayed in https://specification.modelica.org/maint/3.5/operators-and-expressions.html#spatialdistribution

dietmarw, 2021-03-15:
Sorry forgot to address this question proper ly to @HansOlsson and @MartinOtter

HansOlsson, 2021-03-15:
> I was wondering when fixing this if using SVG directly would be allowed already for the next MSL release. This would allow me to include the equations displayed in https://specification.modelica.org/maint/3.5/operators-and-expressions.html#spatialdistribution

I don't know what Modelica Language version we will allow.
Note that Modelica 3.5 allows SVG for bitmaps, but we haven't clarified what is allowed for HTML (and the documentation will be in HTML).

beutlich, 2021-03-15:
I'd also prefer to not include Modelica 3.5 elements as long as it is not clear what version next maint or main versions are based on.

dietmarw, 2021-03-15:
OK that's fine. The other issue I have is how much ModelicaReference should be a copy of the Modelica Specification. The easiest solution would of course just to copy paste the parts from the specification. But this would also mean having to maintaining duplicates. Not sure how sensible this is.

tobolar, 2021-03-16:
getInstanceName is, actually, a nice feature. If would be helpful to have some rule/reccomendation in MSL how to use it properly.

dietmarw, 2021-03-16:
@tobolar  But ModelicaReference is not meant to be a guide line for MSL but rather a lookup for Modelica in general. If you wish some guideline for MSL then that should probably be added under `Modelica.UsersGuide.Conventions`.

tobolar, 2021-03-16:
@dietmarw Definitely! Sorry for mixing it here.

PR #3808 (2021-06-09): getInstanceName and spatialDistribution not listed in ModelicaReference
Fixes #3773 by adding the two missing operators to the list.
commit 7cf4176f (2021-06-09): Adds getInstanceName and spatialDistribution

Fixes #3773

* Add system of equations to be solved by spatialDistribution

Generate image using http://latex2png.com at 120dpi

```tex
\begin{array}{rcl}
\frac{\partial z(y,t)}{\partial t}+v(t)\frac{\partial z(y,t)}{\partial y} & = & 0.0\\
z(0.0, t) & = & \mathrm{in}_0(t) \text{ if $v\geq 0$}\\
z(1.0, t) & = & \mathrm{in}_1(t) \text{ if $v<0$}
\end{array}
```
closed 2021-06-09 by dietmarw