modelica/ModelicaStandardLibrary #3775: Why should function names start with a lower case letter?

HansOlsson, 2021-03-15:
I noticed that we have the general guideline saying that classes start with upper case - _except for functions_:
https://github.com/modelica/ModelicaStandardLibrary/blob/19739745cf3ec63ac00763a355dd479e9d96254b/Modelica/package.mo#L1244

I'm not necessarily proposing to change this, but wanted to discuss how important this is in general (also outside of MSL).

If we actually look at the usage it usually doesn't seem to matter, since package-names start with upper-case so it looks as follows:
```
algorithm 
  I_numerical[1] := Modelica.Math.Nonlinear.quadratureLobatto(
      function Modelica.Math.Nonlinear.Examples.UtilityFunctions.fun4(),
      0,
      1,
      tolerance);
  I_analytical[1] := -cos(1) + cos(0);

  I_numerical[2] := Modelica.Math.Nonlinear.quadratureLobatto(
      function Modelica.Math.Nonlinear.Examples.UtilityFunctions.fun5(w=5),
      0,
      13,
      tolerance);
  I_analytical[2] := -cos(5*13)/5 + cos(5*0)/5;

  I_numerical[3] := Modelica.Math.Nonlinear.quadratureLobatto(
      function Modelica.Math.Nonlinear.Examples.UtilityFunctions.fun6(k=1/
      sqrt(2)),
      0,
      Modelica.Constants.pi/2,
      tolerance);
  I_analytical[3] := 1.8540746773013719184338503;
```

mestinso, 2021-07-16:
...I did put some thought into this recently, and there are two reasons I came up with:
1) Built-in functions(min, max, sin, cos, log, etc, etc.) are lowercase, so it is consistent with that.
2) It's roughly consistent with other object oriented languages: (roughly: Classes are UpperCamelCase, functions, variable names, and class instances are lowerCameCase or snake_case etc). E.g. https://www.python.org/dev/peps/pep-0008/#function-and-variable-names

HansOlsson, 2021-09-28:
> ...I did put some thought into this recently, and there are two reasons I came up with:
>
> 1. Built-in functions(min, max, sin, cos, log, etc, etc.) are lowercase, so it is consistent with that.
> 2. [...]

But my point is that it's not the same as the other languages use member functions, so in Modelica you have `Modelica.Math.Nonlinear.quadratureLobatto` (as package names are in upper case - it's just the last part that is lowerCase) and in Python `self.quadratureLobatto` (as instance names are in lowerCase).

I'm not saying that there's a major problem with lowerCase for function names in Modelica - just that it seems fairly unimportant for most actual use-cases.

sjoelund, 2021-09-28:
> I'm not saying that there's a major problem with lowerCase for function names in Modelica - just that it seems fairly unimportant for most actual use-cases.

It's pretty nice to be able to distinguish things like blocks and records from function calls just by seeing the code.