modelica/ModelicaStandardLibrary #3778: Reference result degradation for TestValvesCompressibleReverse

henrikt-ma, 2021-03-17:
The `ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse` was [briefly commented](https://github.com/modelica/ModelicaStandardLibrary/issues/3415#issuecomment-590975464) in #3415, but maybe nobody actually took a closer look?

Did the dynamics of the model really change from version 3.2.3 in a way that can explain the kind of difference shown below?
[image]

Similar features are seen other variables as well:
- `V3.port_b.m_flow` at time 1.7728
- `SourceP1.ports[2].m_flow` at time 1.8072
- `V2.port_b.m_flow` at time 1.8072
- `V2.port_b.p` at time 1.82
- `V3.port_b.p` at time 1.82

beutlich, 2021-03-17:
As stated in #3415, this models gives
```
ERROR: Failed to solve nonlinear system using Newton solver.
  Time: 1.768
```
with default solver settings. Apparently none of the librarry officers evaluated that failing model and I was happy enough to get it simulating with different settings (and different results as you reported now).

thorade, 2021-03-17:
The deviation seems to be around zero flow condition. Was the handling of zero flow changed? Maybe regularization? Some regRoor or regSquare function updated?
I am not really familiar with this model, but could try to help with investigation

henrikt-ma, 2021-03-17:
I would recommend to start by comparing the 3.2.3 and 4.0.0 results using the same tool and settings.  With System Modeler, I haven't noticed any difference between the 3.2.3 and 4.0.0 results.

beutlich, 2021-03-17:
OK, I figured out what the reason this. TL;DR There is no degradation.

* The simulation results of https://github.com/modelica/MAP-LIB_ReferenceResults/tree/v4.0.0/ModelicaTest/Fluid/TestComponents/Valves/TestValvesCompressibleReverse still can be reproduced with Dymola 2021x and CSV-Compare does not find any relevant deviations w.r.t. this MSL v4.0.0 result.
* Although the simulation run at https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v3.2.3/ModelicaTest/Fluid/TestComponents/Valves/TestValvesCompressibleReverse/creation.txt#L25 reports 5000 sample points, the actual simulation result file https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v3.2.3/ModelicaTest/Fluid/TestComponents/Valves/TestValvesCompressibleReverse/TestValvesCompressibleReverse.csv only contains about 500 sample points. I can imagine how this happend. As stated in https://github.com/modelica/ModelicaStandardLibrary/issues/3415#issuecomment-590975464, the simulation does not succeed with default settings of tolerance and sample points. Thus @GallLeo tried to manually figure out suitable settings that finally lead to a successful simulation. He then manually edited https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v3.2.3/ModelicaTest/Fluid/TestComponents/Valves/TestValvesCompressibleReverse/creation.txt#L26, but forgot to adpt the number of intervals. Thus, this creation.txt does not tell the truth how the MSL v3.2.3 simulation result was created.
* Running the model again in Dymola 2021x (by command `ReleaseChecks.simulateExecutables({"ModelicaTest"}, {"modelica://ModelicaTest/Resources/Reference"}, false, useTolerance=true, tolerance=0.0001, gitRevision="6626538a23", numberOfIntervals=500, gitStatus="M ModelicaServices/package.mo", description="Repeated run for https://github.com/modelica/ModelicaStandardLibrary/issues/3778");`) and comparing to the result from MSL 3.2.3 results in no significant deviations.
* Thus, there is no degradation of model/library or solver quality between MSL 3.2.3 and 4.0.0 in Dymola.
* However, there actually is a problem in the model itself as can be seen in https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0/ModelicaTest/Fluid/TestComponents/Valves/TestValvesCompressibleReverse/simulate_passed.log and that strange simulation result that became visible when comparing result of different quality.

henrikt-ma, 2021-03-18:
So if Dymola 2021x doesn't find any difference between the 3.2.3 and 4.0.0 versions of the model, isn't this reason to believe that the model hasn't changed, but that one of the reference result has inferior quality due to being generated with a different versions of Dymola?

If that's the case, I would be very helpful to see a statement about which reference has the inferior quality, so that I can reference this issue if I need to make strange adaptations to cope with it in our test setup.

beutlich, 2021-03-19:
> Thus, there is no degradation of model/library or solver quality between MSL 3.2.3 and 4.0.0 in Dymola.

OK, I change "or" to "and": Thus, there is no degradation of model/library and solver quality between MSL 3.2.3 and 4.0.0 in Dymola. It is not a tool issue.

Again, it is only due to the number of sample points. If you set 500 sample points (no matter which MSL version or which Dymola version) you will lucky enough to not see a smooth curve for the respective interval. If you set 5000 sample points (no matter which MSL version or which Dymola version) you will see the detailed curve with stepped results for the respective interval.

And yes, it might a good advice that some of the library officers could take a closer look at this model and why it is numerically fragile.

henrikt-ma, 2021-03-19:
> Again, it is only due to the number of sample points. If you set 500 sample points (no matter which MSL version or which Dymola version) you will be lucky enough to see a smooth curve for the respective interval. If you set 5000 sample points (no matter which MSL version or which Dymola version) you will see the detailed curve with stepped results for the respective interval.

Alright, so what I hear is that given the model and generating tool, there has been a potential degradation in the way the reference result has been generated, combined with an incorrect description of how it was generated in 3.2.3.  If the "new" features seen above are thought to be artifacts of numerical fragility, this means to me that a setting of 5000 output intervals is inappropriate for this model and tool.  As long as this becomes documented, it will be much easier to adapt for tool vendors that try to take the reference results seriously.

beutlich, 2021-03-19:
Yes, the MSL 3.2.3 creation.txt is incorrect.

I still see the numerical fragility as a model or library issue and prefer to have that one fixed, such that the model runs with default solver settings.

henrikt-ma, 2021-03-19:
> I still see the numerical fragility as a model or library issue and prefer to have that one fixed, such that the model runs with default solver settings.

If it could really be concluded that the model is fragile, such a conclusion would show that there's no point in trying to reproduce any of the 3.2.3 or 4.0.0 reference results.  Of course, the model fragility should be fixed in this case, rather than making arbitrary changes to the reference results.

However, not all evidence point in this direction.  System Modeler doesn't display this kind of fragility at all, and OpenModelica is also troubled by the new reference result (and as far as I know, they generally try to mimic the simulation settings used for reference creation):
- https://libraries.openmodelica.org/branches/master/ModelicaTest_4.0.0/files/ModelicaTest_4.0.0_ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse.diff.html

Of course, that some tools don't display problems isn't evidence that a model is sound, it only suggests that it might be.

beutlich, 2021-03-19:
Hm, maybe your are right. I also checked in SimulationX and it cannot reproduce the reference result of `SourceP2.ports[1].m_flow`, but rather gives a smooth curve similar as WSM or OMC.

@HansOlsson Can you please check and analyze why Dymola Dassl solver has troubles with the non-linear systems and comes to a solution that cannot reproduced by the three other simulation tools?

@henrikt-ma Would it be OK for you to remove the four "wrong" signals from the ref results for now?

henrikt-ma, 2021-03-19:
> @henrikt-ma Would it be OK for you to remove the four invalid signals from the ref results for now?

I don't think that is for me to tell what is OK or not.  All I need short-term is a clear statement that the 4.0.0 reference seems problematic.  Then I'll just use that to explain why I allow our test to pass despite not showing the new artifacts.

beutlich, 2021-03-19:
> All I need short-term is a clear statement that the 4.0.0 reference seems problematic.

I think you can answer that yourself now.

henrikt-ma, 2021-03-19:
Without a deeper analysis, I can't exclude the possibility that Dymola reveals an issue with the model by applying some valid optimization that other tools don't do.  Therefore, I'll wait for the response from @HansOlsson instead of jumping to conclusions.

HansOlsson, 2021-03-23:
> Hm, maybe your are right. I also checked in SimulationX and it cannot reproduce Dymola's reference result of `SourceP2.ports[1].m_flow`, but rather gives a smooth curve similar as WSM or OMC.
> 
> @HansOlsson Can you please check and analyze why Dymola Dassl solver has troubles with the non-linear systems and comes to a solution that cannot be reproduced by the three other simulation tools?

I tracked it down to a detail of the stream handling, and found that it is possible to modify it and completely avoid the issue for this model. I just have to investigate if that is unproblematic in general or not.

Assuming there's nothing else I would say that any model-issue is due to the stream-handling, and if we should update anything it would to be improve the details for stream-connectors.

beutlich, 2021-05-21:
Fix confirmed with Dymola 2022 (though not mentioned in the [release notes](https://www.3ds.com/fileadmin/PRODUCTS/CATIA/DYMOLA/PDF/Dymola-2022-release-notes.pdf)).

closed 2021-05-21 by beutlich

henrikt-ma, 2021-05-25:
> Fix confirmed with Dymola 2022 (though not mentioned in the [release notes](https://www.3ds.com/fileadmin/PRODUCTS/CATIA/DYMOLA/PDF/Dymola-2022-release-notes.pdf)).

Would it perhaps be more relevant to wait for the reference result to be recreated without the artifacts before closing, than waiting for a fix in Dymola?

beutlich, 2021-05-25:
Can't follow your proposal here. It is a tool-issue, which is fixed now. Whenever a new MSL released, major or maintenance, the fix will be available.

henrikt-ma, 2021-05-25:
From my point of view, this is not a tool issue.  I see it as an issue with the reference result that I don't consider fixed until a new reference result is available on the _master_ of https://github.com/modelica/MAP-LIB_ReferenceResults.