modelica/ModelicaStandardLibrary #3782: What is the quality of the TankWithEmptyingPipe1 reference result?

henrikt-ma, 2021-03-22:
In the same spirit as #3778, I'm wondering about the quality of the reference result for `Modelica.Fluid.Examples.AST_BatchPlant.Test.TankWithEmptyingPipe1`.  This is what `tank1.medium.T` looks like in System Modeler:
[image]

The System Modeler result is quite robust to changes in method and tolerance parameters.

To build confidence for opening an issue about the difference, @maltelenz provided me with the corresponding OpenModelica result (which happens to be treated as correct in [their testing framework](https://libraries.openmodelica.org/branches/master/Modelica_4.0.0/Modelica_4.0.0.html)):
[image]

Although OpenModelica's result is closer to the reference, it is still sufficiently difference to indicate a potential problem.

I'd be grateful if someone could confirm that the reference result is of good quality and worth trying to reproduce with good accuracy.

HansOlsson, 2021-03-23:
From Dymola's perspective it's just a matter of varying the tolerance; with default tolerance we get close to reference and with stricter we get to about the same as the SystemModeler results.

henrikt-ma, 2021-03-23:
OK, good to hear!

Despite the fact that System Modeler doesn't require stricter tolerance to produce this result, this is a typical example of a situation where I would have liked to see a routine that would involve setting a suitable `Tolerance` in the model, a reference being generated with even more strict tolerance, and a verification that the model's tolerance is sufficient to come close to the reference.

beutlich, 2021-03-23:
Looks like we can fix it in the model. @HansOlsson Can you please create a PR setting a better tolerance. Thank you.

beutlich, 2021-05-21:
> Can you please create a PR setting a better tolerance.

Available by #3813.

PR #3813 (2022-10-07): Decrease solver tolerance of Modelica.Fluid.Examples.AST_BatchPlant.Test.TankWithEmptyingPipe1
Resolves #3782.
closed 2022-10-07 by MartinOtter