modelica/ModelicaStandardLibrary #3792: ModelicaStrings.c uses strtol_l and strtod_l under gcc

DagBruck, 2021-04-12:
In ModelicaStrings.c there is conditional code to use the locale-specific functions `strtol_l` and `strtod_l` when available. That is a good idea of course.

However, the test under **GCC** seems to be incomplete; as far as I can tell, the `xx_l` functions are a non-standard extension, and only defined if `defined(__USE_GNU)` is defined. The `#if` tests should be updated?

The compiler messages I get on Ubuntu 20.04 LTS and gcc 9.3.0 are:

```
/mnt/e/Program Files/Dymola 2022 Beta 3/source/ModelicaStrings.c: In function ‘ModelicaStrings_scanInteger’:
/mnt/e/Program Files/Dymola 2022 Beta 3/source/ModelicaStrings.c:327:26: warning: implicit declaration of function ‘strtol_l’; did you mean ‘strtoll’? [-Wimplicit-function-declaration]
327 | x = (int)strtol_l(buf, &endptr, 10, loc);

/mnt/e/Program Files/Dymola 2022 Beta 3/source/ModelicaStrings.c: In function ‘ModelicaStrings_scanReal’:
/mnt/e/Program Files/Dymola 2022 Beta 3/source/ModelicaStrings.c:436:13: warning: implicit declaration of function ‘strtod_l’; did you mean ‘strtoull’? [-Wimplicit-function-declaration]
436 | x = strtod_l(buf, &endptr, loc);

```

beutlich, 2021-04-12:
Duplicate of #3544. OK to close as tool-issue?

DagBruck, 2021-04-12:
> Duplicate of #3544. OK to close as tool-issue?

Actually, I think not. What you say in #3544 is that if you include features.h with the right contents in the right order, then you get the needed declaration. My argument is that because we actually have an `#if` test in ModelicaStrings.c it should be correct and do the right thing if you for some reason want to compile Modelica models as strictly standard-conforming C code. (And with eFMI that is not an option.)

You are of course right that it is a duplicate issue, but it is a valid issue.

beutlich, 2021-04-12:
Does the issue also show up if you include ModelicaStrings.c first in your single compilation unit?

henrikt-ma, 2021-04-12:
> In ModelicaStrings.c there is conditional code to use the locale-specific functions `strtol_l` and `strtod_l` when available. That is a good idea of course.

It doesn't sound like a good idea if it isn't even documented that a function like `Strings.scanReal` may depend on locale.  Further, the documentation for `Strings.Advanced.scanReal` doesn't even seem to leave any room for locale dependence.

As a user, I'd be quite frustrated if it turned out that my `scanReal` was behaving differently than the `scanReal` of a collaborator with a tool built with a different locale.

beutlich, 2021-04-12:
You can build the sources with `NO_LOCALE` in that case.

HansOlsson, 2021-04-12:
> It doesn't sound like a good idea if it isn't even documented that a function like `Strings.scanReal` may depend on locale. Further, the documentation for `Strings.Advanced.scanReal` doesn't even [...]

That's why it's an **amazingly good idea** to use strtol_l instead of strtol - to avoid the locale dependency that the C-standard created.

henrikt-ma, 2021-04-12:
Yes, now I feel deeply amazed.  The reason for using the "locale-specific" functions is really to avoid locale dependency?!  So the insane reason to build your own MSL with `NO_LOCALE` would be to actually get locale dependent behavior?!

(As long as specifying `NO_LOCALE` could only be used to get an implementation not matching the documentation of `Strings.Advanced.scanReal`, I'm fine with its effect not being documented, assuming that anyone specifying `NO_LOCALE` will be a power user who can figure out the consequences without help of the documentation.)

DagBruck, 2021-04-12:
Well, you always have an active locale, but the Modelica behaviour specifically corresponds to the standard C locale, not any localaized locale for Sweden and Germany (for example). But even without `strtol_l` you can get the right behaviour, but it requires you to set/reset the global locale setting or do some other trick.

`scanReal` actually performs such a trick, it replaces any decimal point with the decimal symbol used by the currently active locale (e.g. ",") so as far as I can tell it does mostly the right thing even if `strtol_l` is undefined.

But in any case, my point was that code in MSL should never call a function without including the header file that declares that function, regardless of compilation order. And even less so if it is a non-standard function.

henrikt-ma, 2021-04-12:
I see.  Thanks for the explanation!

beutlich, 2021-04-12:
> 
> 
> Does the issue also show up if you include ModelicaStrings.c first in your single compilation unit?

@DagBruck Can you confirm that the warning is gone then?

beutlich, 2021-04-16:
Closing as (duplicate) tool-issue.

closed 2021-04-16 by beutlich

DagBruck, 2021-04-16:
There is a solution that only relies on Standard C. As far as I can tell it will solve the problem without any library extensions and conditional compilation.

```
const char* old_locale = setlocale(LC_NUMERIC, "C");
x = strtod(buf, &endptr);
setlocale(LC_NUMERIC, old_locale);
```

HansOlsson, 2021-04-16:
> There is a solution that only relies on Standard C. As far as I can tell it will solve the problem without any library extensions and conditional compilation.
> 
> ```
> const char* old_locale = setlocale(LC_NUMERIC, "C");
> x = strtod(buf, &endptr);
> setlocale(LC_NUMERIC, old_locale);
> ```

However, with bad performance, and with unpredictable results in multi-threaded programs. The most annoying part is that we get the performance penalty even in the C-locale.