modelica/ModelicaStandardLibrary #3814: Unify spelling of controllers

christiankral, 2021-05-22:
In `Modelica.Blocks.Continuous` the controllers are spelled, for example, the following way:

 - PI-controller
 - PID-controller

In Electropedia the following spelling is used:

- [PI element](https://www.electropedia.org/iev/iev.nsf/display?openform&ievref=351-50-11) or *proportional plus integral element*
- [PID element](https://www.electropedia.org/iev/iev.nsf/display?openform&ievref=351-50-19) or *proportional plus integral plus derivative element*

From my understanding it would thus also makes sense to omit the dash when using the term *controller* instead of *element*:

- PI controller or *proportional plus integral controller*
- PID controller or *proportional plus integral plus derivative controller*

A comment from a native English speaking person is highly appreciated here. Possibly @mtiller or @harmanpa could comment here.

Once decided, this shall be documented in `Modelica.UsersGuide.Conventions.Terms.Electrical`.

AHaumer, 2021-05-26:
Personally, I dislike the term "PI element", as well as "_proportional plus integral controller_". 
What's the difference between "PI-controller" and "PI controller"?
For me, "PI-controller" is somehow easier to read. Maybe we shouldn't look at Wikipedia, but into a textbook about control theory written by a antive speaker.

christiankral, 2021-05-27:
I do not focus in the term *element*. To me this ticket is simply a matter of using the dash or not: *PI-controller* or *PI controller*. I somehow guess that the dash is not correct and I base my suspicion on Electropedia.

I makes, however, sense to unify the spelling all over the MSL and add the correct spelling to [Modelica.UsersGuide.Conventions.Terms.Electrical](https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_UsersGuide_Conventions_Terms.html#Modelica.UsersGuide.Conventions.Terms.Electrical).

christiankral, 2021-05-27:
So this issue needs input from a native English speaker please.

christiankral, 2021-09-23:
@mtiller Could you possibly help here?

christiankral, 2022-07-04:
@harmanpa @mtiller Could you please comment on this spelling issue?

harmanpa, 2022-07-04:
I would write "PI controller" / "PID controller"

christiankral, 2022-07-05:
Thanks @harmanpa, this makes sense. We shall omit the dash then.

dietmarw, 2022-09-21:
[Grammerly](https://www.grammarly.com) agrees with @harmanpa on this. So we should implement it.

TManikantan, 2023-05-05:
I found out the location at which these changes should be made
1. `Modelica.Mechanics.MultiBody.Examples.Systems.RobotR3.Utilities.Controller`

TManikantan, 2023-05-05:
I have short listed the instances to which changes have to be made.
1.`Modelica.Mechanics.MultiBody.Examples.Systems.RobotR3.Utilities.Controller` (changes in documentation).
2.`Modelica.Electrical.Machines.Utilities.DQCurrentController` (changes in documentation).
3.`Modelica.Electrical.Machines.Examples.ControlledDCDrives.Utilities.LimitedPI` (changes in class description).
4.`Modelica.Blocks.Continuous.PID` (changes in Documentation and class Description).

Other than these i noticed that some description or documentation also carry spelling "**_P-controller_**" . 

1.`Modelica.Mechanics.MultiBody.Examples.Systems.RobotR3.Utilities.Controller` (Documentation).
2.`Modelica.Electrical.Machines.Examples.ControlledDCDrives.Utilities.LimitedPI` (Documentation).
3. `Modelica.Blocks.Continuous.LimPID` (Documentation) .
4.  `Modelica.Clocked.Examples.Systems.ControlledMixingUnit` (Documentation).

Should i go ahead make changes to above mentioned instances? If yes, Should i also change **_P-controller_** to **_P controller_** ?

tobolar, 2023-05-23:
> I have short listed the instances to which changes have to be made. ...

What about
- "OnOff-controller" in `Thermal.HeatTransfer.Examples.ControlledTemperature` and
- "Voltage-Frequency-Controller" in `Electrical.Machines.Utilities.VfController` and `Magnetic.QuasiStatic.FundamentalWave.Utilities.VfController`?

(Btw. do we need both "VfControllers"? They seem to be quite similar!)

christiankral, 2023-05-23:
> What about
> 
> * "OnOff-controller" in `Thermal.HeatTransfer.Examples.ControlledTemperature` and
> * "Voltage-Frequency-Controller" in `Electrical.Machines.Utilities.VfController` and `Magnetic.QuasiStatic.FundamentalWave.Utilities.VfController`?

They also need different spelling.

PR #4136 (2023-06-15): Unify spelling of controllers
Close #3814. unify spelling of controllers by omitting dash in description and documentation
closed 2023-06-15 by MartinOtter