modelica/ModelicaStandardLibrary #3816: Remove Modelica.Electrical.Machines.Examples.ControlledDCDrives.Utilities.Battery

christiankral, 2021-05-22:
As `Modelica.Electrical.Machines.Examples.ControlledDCDrives.Utilities.Battery` is located in an Examples package, I guess it is save to move this model to `ObsoleteModelica4.mo`, as maintenance is not guaranteed for models located within Examples. I think this model is solely used in `Modelica.Electrical.Machines.Examples.ControlledDCDrives.Utilities.PartialControlledDCPM` and can be replaced by `Modelica.Electrical.Batteries.BatteryStacks.CellStack`.

dietmarw, 2021-05-25:
Removal from within Examples should be not a problem. We normally don't keep examples inside `ObsoleteModelica*` so I would simply drop it. But what might be a problem here is that the `Utilities.Battery` is a nice simple ideal battery whereas the `CellStack` is a rather more complicated (and realistic) representation that for some use-cases might simply be to complicated. 
So what about adding a similar ideal battery model next to CellStack and use that in the example. The old one could be simply be dropped or an conversion rule added (though not really necessary).

AHaumer, 2021-05-26:
Hmmm ... this very simple model doesn't really fit into Batteries, since it's just a constant voltage with inner resistance, I'ts more a linear voltage source. It's named Battery only to have a "realistic feeling" of the examples. Maybe we shold rename it to "Linear voltage source"? But then why not introduce a LinearVoltageSource (with inner serial resistor) and a LinearCurrentSource (with inner parallel conductor) to Electrical.Analog.Sources?

dietmarw, 2021-05-26:
How about simply replacing that model with a voltage source with a series resistance in the example and removing the `Utilities,Battery` model altogether.

closed 2022-07-26 by christiankral