modelica/ModelicaStandardLibrary #3820: skipWhiteSpace under-documented

DagBruck, 2021-05-28:
The function Modelica.Utilities.Strings.Advanced.skipWhiteSpace appears to be under-documented. The cases I have found are:
1. What does it return when the given string contains only whitespace or is an empty string?
2. What does it return if the start index is greater than the string length.

Here are three test cases.

```
Modelica.Utilities.Strings.Advanced.skipWhiteSpace("")
Modelica.Utilities.Strings.Advanced.skipWhiteSpace("  ")
Modelica.Utilities.Strings.Advanced.skipWhiteSpace(" a ", 700)

```

DagBruck, 2021-05-28:
> ```
> Modelica.Utilities.Strings.Advanced.skipWhiteSpace("")
> Modelica.Utilities.Strings.Advanced.skipWhiteSpace("  ")
> Modelica.Utilities.Strings.Advanced.skipWhiteSpace(" a ", 700)
> ```
 Might add that it returns 1, 4 and 700 for these cases. The last one in particular is questionable.

sjoelund, 2021-05-28:
> > ```
> > Modelica.Utilities.Strings.Advanced.skipWhiteSpace("")
> > Modelica.Utilities.Strings.Advanced.skipWhiteSpace("  ")
> > Modelica.Utilities.Strings.Advanced.skipWhiteSpace(" a ", 700)
> > ```
> 
> Might add that it returns 1, 4 and 700 for these cases. The last one in particular is questionable.

Is that a Windows thing? It returns 1, 3 and 700 for me on Linux...

DagBruck, 2021-05-28:
> > Might add that it returns 1, 4 and 700 for these cases. The last one in particular is questionable.
> 
> Is that a Windows thing? It returns 1, 3 and 700 for me on Linux...

No a typo on my part :-)