modelica/ModelicaStandardLibrary #3836: Incorrect results from isPowerOf2 for large inputs

henrikt-ma, 2021-06-10:
Because of the `Real` arithmetic going on inside `Modelica.Math.isPowerOf2`, it will produce incorrect results (erring on the side of returning `true` more often than it should) for inputs outside the range than can be represented exactly as a `Real`:
```
  result := i == 2 ^ integer(log(i) / log(2) + 0.5);
```

It can be noted that the problem here is not in the `Real` expression `2 ^ integer(log(i) / log(2) + 0.5)`, since it is a power of 2.0, but in the implicit coercion of `i` to `Real` on the other side of the equality.  Hence, I think this should be sufficient to fix the problems:
```
   result := i == integer(2 ^ integer(log(i) / log(2) + 0.5)); /* Use comparison on Integer values, not Real. */
```

One could also argue that the problem is that the language doesn't provide a function for raising an `Integer` to a non-negative `Integer` power, but for now `isPowerOf2` needs to work with what is available.

dietmarw, 2021-06-10:
Library Officers for Math are as assigned now (see also https://github.com/modelica/MAP-LIB_ProjectRules/blob/master/MAP-LIB_ProjectRules.md#annex-3-library-officers)

HansOlsson, 2021-06-10:
I added a pull-request with a slightly safer variant of it.
However, I don't see the failure-mode - the input `i` is an integer, so if the right-hand-side is too larger, how can it be equal to the integer?

henrikt-ma, 2021-06-10:
The secret is that the right hand side can have the exact value of a power of 2, but at a precision that is generally insufficient to handle integers.  However, when converted to an integer type, one gets the desired `Integer`.

HansOlsson, 2021-06-10:
> The secret is that the right hand side can have the exact value of a power of 2, but at a precision that is generally insufficient to handle integers. However, when converted to an integer type, one gets the desired `Integer`.

As stated in the PR that does not make sense to me, if the reals don't suffice to handle the integer it doesn't help to cast it to integer - as the precision was already lost. We can, of course, handle it.

HansOlsson, 2021-06-11:
Thinking more I can see that the first proposal might make sense when looking at exact details of how floating point numbers are represented, but it will require a quite detailed explanation. However, the PR I constructed all that and instead relies on a power of two being `2^(2*k+1)` and checking that` i/2^k = 2^(k+1)` and `i%(2^k) == 0`, thus avoiding all of the large numbers.

PR #3843 (2021-09-09): Fully integer based version.
Closes #3836 
This avoids floating point math and also overflow.
It actually seems to be a bit faster.
closed 2021-09-09 by MartinOtter