modelica/ModelicaStandardLibrary #3841: Include function definition in parameter record legal?

AHaumer, 2021-06-13:
Is it legal Modelica to include a function definition in a parameter record?
```
record Data "Material parameters"
  extends Modelica.Icons.Record;
  parameter Real x1 "Nominal point x";
  parameter Real y1 "Nominal pint y";
  parameter Real s0 "Slope beyond nominal point";
  function approx
    extends Modelica.Icons.Function;
    input Data data "Materal parameters";
    input Real x "Actual x";
    output Real y "Actual y";
  algorithm
    y:=data.y1/data.x1*min(x, data.x1) + data.s0*max(x - data.x1, 0);
  end approx;
end Data;
```
This would open the possibility to use different approximations for material behaviour just by choosing the material, 
if the approximation functions for different materials have the same name and structure (inputs and output).

beutlich, 2021-06-14:
Functions in records are used in [ExternData](https://github.com/modelica-3rdparty/ExternData), too. Note, that they need to be public.

AHaumer, 2021-06-16:
Ok thanks a lot @beutlich ! I have to admit it's a bit more complicated:
I want to choose from different materials, and these are grouped by approximation method of a certain property. 
The property is the same for all materials. For each group there's a base record, defining the parameters and the approximation function (which uses the parameter record!). I can't get a legal solution, - well hard to explain, easier to look at: I've enclosed what I have so far. Any hints welcome!
[TestLib.zip](https://github.com/modelica/ModelicaStandardLibrary/files/6663193/TestLib.zip)

AHaumer, 2021-06-17:
Analyzing the problem:

- Common base parameter record to be able to choose from different parameters sets (with different approximation function)
- Common base approximation function to use it in the components, independent of the later chosen materia
- Use of the parameter set as input to the approximation function 
- Approximation function included in the parameter record (tied to the parameters)
- Collections of parameter sets, each collection witg different parameters and different approximation function

closed 2021-11-08 by beutlich