modelica/ModelicaStandardLibrary #3844: Conflicting guidelines for HTML documentation convention guidelines for tables

dietmarw, 2021-06-18:
In `Modelica.UsersGuide.Conventions.Documentation.Format.Tables` the guidelines say that the caption should be defined as:

```
<caption align="bottom"><strong>Tab 2:</strong> Caption starts with a capital letter</caption>
```
But this would place the caption at the **bottom** and not the top. The problem is that Dymola renders the position incorrectly and always displays it on the top. Now the question, should the caption appear at the top of the table or the bottom. If at the top then it needs to be updated and tools need to make sure to get the rendering right.

One additional problem is that `align` is no longer supported in HTML5, so the guideline should most likely updated  anyway.

christiankral, 2021-06-26:
I assume we do not have a decision yet, if HTML5 is the standard that the MLS and the MSL rely on, see https://github.com/modelica/ModelicaSpecification/issues/2395. 

However, we could 
1. remove `align` from the MSL documentation
```modelica
<caption><strong>Tab 2:</strong> Caption starts with a capital letter</caption>
```
2. and do not specify where the caption shall be placed and thus leave it up to the tool vendors how they implement it

dietmarw, 2021-06-28:
I agree that is the most practical and pragmatic solution and also future proof. I'll create a PR to remove it.

PR #3848 (2021-07-01): Remove `align=bottom` from guidelines and code
This fixes #3844 by removing the recommendation and application of `align=\"bottom\"` since tools tend to interpret this differently (e.g., ignore) and the `align` parameter is also deprecated in HTML5. Not using and recommending it  makes it more future proof.
commit 1f15f149 (2021-07-01): Remove `align=bottom` from guidelines and code

This fixes #3844 by removing the recommendation and application of `align=\"bottom\"` since tools tend to interpret this differently (e.g., ignore) and the `align` parameter is also deprecated in HTML5. Not using and recommending it  makes it more future proof.
closed 2021-07-01 by christiankral