modelica/ModelicaStandardLibrary #3860: CO2 normalBoilingPoint = -1

jansilar, 2021-08-24:
Hi, may be I miss something, but why is the CO2 normalBoilingPoint = -1 in Modelica.Media.IdealGases.Common.FluidData?
I think the correct value is 194.67°K (https://www.thermopedia.com/content/613/)
The Modelica.Media.Interfaces.Types.Temperature type has min=1, so that it is out of the range.

HansOlsson, 2021-08-24:
I'll let one of the experts answer on exactly why that value is used; and what the implications are. But I can understand that it is more complicated as technically CO2 doesn't boil in the sense of going from liquid to gas at that point, but it sublimes from solid to gas; and the media-library is focused on liquids and gases.

bilderbuchi, 2021-09-04:
..at normal pressure, that is. Between ~5 and ~70 bar, you can have a liquid-gas phase transition iirc.

hubertus65, 2021-12-03:
It would be nice to have something like a symbolic constant for "Undefined" or "NaN" here, which doesn't exist in Modelica. The -1 was my workaround to that. Unless somebody has a much better idea, I propose to close it.

bilderbuchi, 2021-12-03:
NaN really sounds useful, yeah. Inf, too, also with an eye on ieee754. See also https://github.com/modelica/ModelicaSpecification/issues/2048 https://github.com/modelica/fmi-standard/issues/644

jansilar, 2023-10-03:
It seems -1 is currently the best solution. I am closing the task. Sorry for the two-year delay :(.

closed 2023-10-03 by jansilar