modelica/ModelicaStandardLibrary #3864: Undesired minimum gain of LimPID

christoff-buerger, 2021-09-13:
`Modelica.Blocks.Continuous.LimPID` has `min=0` for the controller gain `k`. I wonder if that is a good idea and indeed intentional.

I stumbled upon this on a controller developed by DLR, which was more or less a copy and paste of the MSL `LimPID`, but it had a negative gain. Switching to the MSL `LimPID` now caused some strange error, because in the generated code we used the min as a means to saturate parameter changes when recalibratingm causing division by 0 (negative gain saturated to 0).

My issue is, that this default constraint is very questionable:
 1. It is not documented in `LimPID`.
 2. I am not sure this is a good idea. Why do we restrict negative gains in a general component library?
 3. 0 gain is the really unwanted case but we can not easily express it in Modelica (everything except 0, i.e., `]0[`)
 4. At the moment `k=0` is valid, although it will cause `gainTrack.k` to fail due to `gainTrack.k=1(k*Ni)`. So the `k(min=0)` is misleading that 0 is a valid setting.

christoff-buerger, 2021-09-13:
Minor note: This is the bug report related to the issues detected by https://github.com/modelica/efmi-testcases/pull/12

christoff-buerger, 2021-11-30:
@AHaumer, @MartinOtter: Can you have a look into this?

MartinOtter, 2023-03-29:
Yes, you are right "min=0" should be removed for k.

henrikt-ma, 2023-04-28:
@christoff-buerger, could you please edit the initial comment above to include the missing `/` in the expression `gainTrack.k=1(k*Ni)`?

christoff-buerger, 2023-05-02:
> @christoff-buerger, could you please edit the initial comment above to include the missing `/` in the expression `gainTrack.k=1(k*Ni)`?

@henrikt-ma Done

closed 2023-06-01 by TManikantan