modelica/ModelicaStandardLibrary #3865: Why are there no single phase quasistatic RMS voltage and current sensors?

mestinso, 2021-09-14:
The multiphase quasistatic electrical library includes RMS sensors, however they are absent from the single phase quasistatic sensors. I would comment they are generally useful for me and I don't see a good reason why they shouldn't be included. Am I missing anything?

AHaumer, 2021-09-15:
Well, you're welcome to propose an implementation / pull request ...

mestinso, 2021-09-15:
@AHaumer Sure, will plan on it. I wanted to makes sure I wasn't missing something first. Thanks.

christiankral, 2021-09-16:
Reffering to MSL 4.0.0: 

The single-phase sensor `Modelica.Electrical.QuasiStatic.SinglePhase.Sensors.CurrentSensor` senses the RMS current, which is equal to the magnitude of the current `i` flowing through the sensor. 

The polyphase sensor `Modelica.Electrical.QuasiStatic.Polyphase.Sensors.CurrentSensor` senses `m` RMS currents, each of them calculated as the magnitude of the phase currents `i[:]`. In case of a fully symmetrical polyphase system, all the currents `i[:]` are equal to each other. 

There is, however, the additional sensor `Modelica.Electrical.QuasiStatic.Polyphase.Sensors.CurrentQuasiRMSSensor` which calculates the quadratic average phase currents, named as "quasi RMS current": 

`y = sum({sqrt(i[k].re^2 + i[k].im^2) for k in 1:m})/m` 

In case of a fully symmetrical polyphase system the quasi RMS current `y` is equal to each of the phase  currents `i[:]`. 

From my understanding the desired sensors already exists.

mestinso, 2021-09-16:
@christiankral Which component are you specifically referring to when you say the desired sensor already exists? The output of `Modelica.Electrical.QuasiStatic.SinglePhase.Sensors.CurrentSensor` is a complex value (not desired). Also, while the math inside `Modelica.Electrical.QuasiStatic.Polyphase.Sensors.CurrentQuasiRMSSensor` is what i want for m=1, the ports aren't compatible with single phase and the following underlying component doesn't like m=1: `Modelica.Electrical.QuasiStatic.Polyphase.Blocks.QuasiRMS`.

christiankral, 2021-09-16:
OK, now I understand what you mean. 

You are right, the output is always a complex quantity. In the current implementation one has to use the block `Modelica.ComplexBlocks.ComplexMath.ComplexToPolar` to determine the magnitude (RMS value) of a single-phase current.

So please provide a pull request with additional sensors:

- `Modelica.Electrical.QuasiStatic.SinglePhase.Sensors.CurrentRMSSensor`
- `Modelica.Electrical.QuasiStatic.SinglePhase.Sensors.VoltageRMSSensor`

closed 2021-11-18 by MartinOtter