modelica/ModelicaStandardLibrary #3866: startBackward and startForward should not be reference results

uschna, 2021-09-15:
For example in Mechanics.MultiBody.Examples.Systems.RobotR3.OneAxis 
axis.gear.bearingFriction.startBackward
and
axis.gear.bearingFriction.startForward
are reference results.
In
Modelica.Mechanics.Rotational.Interfaces.PartialFriction:
```
		startForward = pre(mode) == Stuck and (sa > tau0_max/unitTorque or pre(
		  startForward) and sa > tau0/unitTorque) or pre(mode) == Backward and
		  w_relfric > w_small or initial() and (w_relfric > 0);
		startBackward = pre(mode) == Stuck and (sa < -tau0_max/unitTorque or pre(
		  startBackward) and sa < -tau0/unitTorque) or pre(mode) == Forward and
		  w_relfric < -w_small or initial() and (w_relfric < 0);
		locked = not free and not (pre(mode) == Forward or startForward or pre(
		  mode) == Backward or startBackward);
...
		mode = if free then Free else (if (pre(mode) == Forward or pre(mode) ==
		  Free or startForward) and w_relfric > 0 then Forward else if (pre(mode)
		   == Backward or pre(mode) == Free or startBackward) and w_relfric < 0
		   then Backward else Stuck);
```
Hence, startForward and startBackward depend from the value of w_relfric in
pre(mode) == Stuck
and the hysteresis in zero-functions. That's why,
startBackward and startForward 
should not be reference results.

henrikt-ma, 2021-09-15:
So what would this imply for `mode`, `free`  and `locked`, which all seem to have a tight connection with `startForward` and `startBackward`?  In my experience, all of these tend to be problematic when it comes to matching the reference results.

HansOlsson, 2021-09-15:
> For example in Mechanics.MultiBody.Examples.Systems.RobotR3.OneAxis
> axis.gear.bearingFriction.startBackward
> and
> axis.gear.bearingFriction.startForward
> are reference results.
> In [...]

I cannot see how that follows.

 If we look closely we see that it `w_small` is 1e10 so `w_relfric` is not above `w_small`, and after the simulation starts `initial()` is false, so `startForward `is not influenced by `w_relfric `after the simulation starts.

uschna, 2021-09-16:
> So what would this imply for `mode`, `free` and `locked`, which all seem to have a tight connection with `startForward` and `startBackward`? In my experience, all of these tend to be problematic when it comes to matching the reference results.

`mode` is for a stable model not the problem. The change of `mode` has only a small time-difference. I think, this is analogously for `free` and `locked`. The problem of `startForward` and `startBackward` is, that they can become 1 = true in a short time. If  `mode` changes from `Stuck ` into `Forward` or `Backward` in the same event-iteration in the reference results, the test is invalid.

uschna, 2021-09-16:
> If we look closely we see that it `w_small` is 1e10 so `w_relfric` is not above `w_small`, and after the simulation starts `initial()` is false, so `startForward `is not influenced by `w_relfric `after the simulation starts.

That is not, what I mean. If `mode = Stuck` in a time and 
`sa > tau0_max/unitTorque`
then
`startForward = true`.
If 
`w_relfric > 0` (1e-26 is enough without hysteresis)
then
`mode = Forward` 
in the same event-step. In the next event-step is
`pre(mode) == Forward <> Stuck`.
Hence, 
`startForward = false`.
There is no 1 = true in the reference results.
But if
`w_relfric == 0` (small numerical difference to 1e-26)
`mode = Stuck`
and we need a small time-step with
`startForward = true` and `mode = Stuck`.
After this time-step `mode` becomes `Forward`.

HansOlsson, 2021-09-16:
Ok, so the problem is only in mode-equation, and the result is that mode is just shifted a bit in time, whereas startForward is somewhat unpredictable - but with small "integral".

I can understand removing startForward and startBackward, since they are false outside transients that aren't important for the reference (so if we just integrate the error we will get zero regardless of whether it matches the reference or not) - but keep mode, free, etc.

beutlich, 2021-09-16:
@uschna Thank you for your analysis. It would be helpful if you could remove the corresponding signal names from the example and test models and create a pull request:
```
Modelica/Resources/Reference/Modelica/Mechanics/MultiBody/Examples/Systems/RobotR3/OneAxis/comparisonSignals.txt
Modelica/Resources/Reference/Modelica/Mechanics/Rotational/Examples/CoupledClutches/comparisonSignals.txt
Modelica/Resources/Reference/Modelica/Mechanics/Rotational/Examples/Friction/comparisonSignals.txt
Modelica/Resources/Reference/Modelica/Mechanics/Rotational/Examples/HeatLosses/comparisonSignals.txt
Modelica/Resources/Reference/Modelica/Mechanics/Rotational/Examples/LossyGearDemo2/comparisonSignals.txt
Modelica/Resources/Reference/Modelica/Mechanics/Rotational/Examples/SimpleGearShift/comparisonSignals.txt
Modelica/Resources/Reference/Modelica/Mechanics/Translational/Examples/Brake/comparisonSignals.txt
Modelica/Resources/Reference/Modelica/Mechanics/Translational/Examples/Friction/comparisonSignals.txt
Modelica/Resources/Reference/Modelica/Mechanics/Translational/Examples/HeatLosses/comparisonSignals.txt
ModelicaTest/Resources/Reference/ModelicaTest/Rotational/AllComponents/comparisonSignals.txt
ModelicaTest/Resources/Reference/ModelicaTest/Rotational/ForUsersGuide/SupportTorque1/comparisonSignals.txt
ModelicaTest/Resources/Reference/ModelicaTest/Rotational/TestBearingConversion/comparisonSignals.txt
ModelicaTest/Resources/Reference/ModelicaTest/Rotational/TestFriction/comparisonSignals.txt
```

uschna, 2021-09-21:
> @uschna Thank you for your analysis. It would be helpful if you could remove the corresponding signal names from the example and test models and create a pull request:
@beutlich: With some problems and some help done. #3868 
@HansOlsson, @henrikt-ma: OK?

henrikt-ma, 2021-09-21:
> @HansOlsson, @henrikt-ma: OK?

I'm not sufficiently familiar with the details of this state machine to have any specific problems to report for the other variables, I just know that they often don't match the reference results precisely, without affecting the overall simulation result.

HansOlsson, 2021-09-21:
> @HansOlsson, @henrikt-ma: OK?

Yes, that makes sense.

uschna, 2021-09-21:
> I'm not sufficiently familiar with the details of this state machine to have any specific problems to report for the other variables, I just know that they often don't match the reference results precisely, without affecting the overall simulation result.

@henrikt-ma: How I described

> The change of `mode` has only a small time-difference. I think, this is analogously for `free` and `locked`.

It is not important that these variables match the reference results precisely. Because of https://github.com/modelica-tools/csv-compare this is not the problem. The problem is only if some values are not near of the reference results.
Moreover, I wrote
> `mode` is for a stable model not the problem.

For instable problems `mode` can jump between `Forward` and `Backward` for numerical random values of `w_relfric`. Did you mean this? But these cases are specific for the model and not the problem of this issue.

henrikt-ma, 2021-09-21:
> @henrikt-ma: How I described
>
>
> It is not important that these variables match the reference results precisely. Because of https://github.com/modelica-tools/csv-compare this is not the problem. The [...]

We don't use csv-compare.  One of the main reasons is that it doesn't offer clear separation of changes at events and what is going on between events.  This might well be the explanation why we find these variables problematic, while other tool vendors don't see a problem.

> Moreover, I wrote
> 
> > `mode` is for a stable model not the problem.
> 
> For instable problems `mode` can jump between `Forward` and `Backward` for numerical random values of `w_relfric`. Did you mean this? But these cases are specific for the model and not the problem of this issue.

I'm not sure, and I don't think I could tell the difference right now between good and bad models in this regard.  Therefore, I suggest that you proceed with this work without my explicit endorsement of the changes.

uschna, 2021-09-21:
> We don't use csv-compare. One of the main reasons is that it doesn't offer clear separation of changes at events and what is going on between events. This might well be the explanation why we find these variables problematic, while other tool vendors don't see a problem.

@henrikt-ma: This is not a problem of csv-compare but a problem of the type of protocolling. If you protocol before and after events you see the changes at events. If the tolerance in time is small enough (only changes in the time-points because of the different solvers), this is only accepted if the change is in the event.
> Therefore, I suggest that you proceed with this work without my explicit endorsement of the changes.

OK, @beutlich I need another reviewer.

beutlich, 2021-09-21:
@uschna Done. Make sure to also sign the CLA (electronically).

uschna, 2021-09-21:
@beutlich Thank you. I discuss it with some colleagues you known. They say that I cannot sign the CLA (electronically) because of
"In order to avoid any conflict of interest situation, the Modelica Association will require from contributors that are employed with a Corporation that generates revenue from Modelica technology to make contributions as employees of that Corporation only and not as individual contributors."
and
"Schedule B: Initial list of designated employees".
This is some difficult.

beutlich, 2021-09-21:
As ESI ITI already signed the CLA, that should not be an issue to worry.

uschna, 2021-09-21:
This we also discuss:
The problem is that the Modelica Association don't know, that I an employee of ESI ITI GmbH, and we don't find an easy way to tell this.
Hence, we don't know what I can do to solve the problem without to inform the manager and the IP Legal team.

henrikt-ma, 2021-09-21:
> @henrikt-ma: This is not a problem of csv-compare but a problem of the type of protocolling. If you protocol before and after events you see the changes at events. If the tolerance in time is small [...]

This sounds interesting, but I don't understand what _protocol_ means in this context.  Is there any written resource where I can learn more about this?

uschna, 2021-09-22:
@beutlich: "This pull request will squash and merge automatically when all requirements are met."
If the second reviewer Martin Otter accept this, will the pull request end, because ESI ITI GmbH already signed the CLA or does ESI need to inform the Modelica Association, that I an employee of ESI ITI GmbH?
Thank you.

beutlich, 2021-09-22:
It will only merge automatically, if you sign the CLA electronically (and a second reviewer approves).

uschna, 2021-09-22:
> This sounds interesting, but I don't understand what _protocol_ means in this context. Is there any written resource where I can learn more about this?

I don't know, if there is any written resource with description of the types of protocolling. If you write a simulation program for Modelica models you can choose which results you protocol (for example equidistant, after all valid steps, (before and) after events, in all event steps and so on).

uschna, 2021-09-22:
@beutlich: Auf Deutsch (ist für mich einfacher): 
Es gab den Vorschlag aus unserer Gruppe, dass ich meinen Pull Request zurückziehe und jemand anders ihn einreicht, der schon registriert ist (Gerd?). Ich habe auch nicht verstanden, was Du mit
"if you sign the CLA electronically"
genau meinst:
https://cla-assistant.io/modelica/ModelicaStandardLibrary?pullRequest=3868:
"Please complete and sign, then scan and email a pdf file of this Agreement to board@modelica.org"?
Dann muss das wirklich über höhere Vorgesetzte und das IP Legal team gehen.
Was schlägst Du vor?
Danke.

beutlich, 2021-09-22:
> 
> 
> @beutlich: Auf Deutsch (ist für mich einfacher):
> Es gab den Vorschlag aus unserer Gruppe, dass ich meinen Pull Request zurückziehe und jemand anders ihn einreicht, der schon registriert ist (Gerd?). Ich habe auch nicht verstanden, was Du mit
> "if you sign the CLA electronically"
> genau meinst:
> https://cla-assistant.io/modelica/ModelicaStandardLibrary?pullRequest=3868:
> "Please complete and sign, then scan and email a pdf file of this Agreement to [board@modelica.org](mailto:board@modelica.org)"?

Siehe https://github.com/modelica/ModelicaStandardLibrary/blob/master/CONTRIBUTING.md: There is no need to scan and send any documents by mail. (if signed electronically).

> Dann muss das wirklich über höhere Vorgesetzte und das IP Legal team gehen.
> Was schlägst Du vor?
> Danke.

Ich werde den Commit-Autor auf mich setzen.