modelica/ModelicaStandardLibrary #3867: Is ModelicaUtilities.h tool-specific?

casella, 2021-09-21:
ModelicaUtilities.h is mentioned in the language specification, [Sect. 12.9.6](https://specification.modelica.org/maint/3.5/functions.html#utility-functions), and available in [Modelica/Resources /C-Sources](https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Resources/C-Sources).

Is it expected to be customized by each tool vendor, or should it be left as on the MA repo?

What is the recommended way to #include it in a .c source code file within a certain library's Resources directory?

sjoelund, 2021-09-21:
If the header was compiler-specific with for example macros that renamed the function into something else, libraries compiled against the generic header might not work in the compiler. So it should not be customized too much (perhaps some changes are needed for varargs on certain platforms, etc... but in general it should be very similar).

henrikt-ma, 2021-09-21:
I see a potential problem with poor separation of specification and standard library here.  Especially in times when the standard library doesn't stay up-to-date with developments in the specification, it is clear that the two should not have mutual dependencies.  That is, the standard library may obviously rely on the specification, while the specification should avoid making statements about the standard library.

Regarding _ModelicaUtilities.h_ it can be debated whether the specification is really making a statement about the standard library.  After all, it just gives the name of the header where these functions should be declared, so the question is whether tools should just consider the _ModelicaUtilities.h_ that comes with the MSL as an example, and that a tool should maintain its own file in synch with current language version, and perhaps also get rid of the one in _Modelica/Resources/C-Sources_ to ensure that it doesn't cause confusion about what is really being included (the one corresponding to the current language version should anyway be placed earlier in the include path to ensure the one from MSL isn't used).

casella, 2021-09-22:
My requirement is to ensure maximum portability of libraries containing external C code across tools.

I'm fine with tools having their own versionof `ModelicaUtilities.h`, based on their internal technology and use of C code, _as long as they don't change the way the functions defined in the spec are called by the third-party external C code_. In this setup, a third-party library developer can just write `#include "ModelicaUtilities.h"` in their C source code, and be sure the appropriate one will be picked in all tools.

I understand this is what Dymola and OpenModelica currently do.

I would just add a note in the specification making this explicit and recommended, so that people writing their own libraries with external C source code know exactly what to do to ensure portability, i.e., just add  `#include "ModelicaUtilities.h"`.

beutlich, 2021-09-22:
> I would just add a note in the specification

We can transfer the issue to the appropriate repository then if you agree.

sjoelund, 2021-09-22:
Well... If there is no need for ModelicaUtilities.h in the resources, perhaps we could move it in MSL as well (to CI files, etc if needed)

beutlich, 2021-09-22:
It still is needed since I build the libs locally on my machine with VS2005.

[image]

henrikt-ma, 2021-09-22:
While it is fine for a library to need this file, each library should not contain the file.  The file should be provided by the Modelica tool environment, which should provide a file corresponding to the currently active language version.

In my opinion, the file should be removed from the MSL to make this clear, and the issue should stay in this repository.

sjoelund, 2021-09-22:
> It still is needed since I build the libs locally on my machine with VS2005.

Yes, but if it's only for compilation it could be higher up in the structure. It doesn't have to be under Resources. Like under https://github.com/modelica/ModelicaStandardLibrary/tree/master/.CI

beutlich, 2021-09-23:
Since the .CI directory is not distributed with the released MSL, see

https://github.com/modelica/ModelicaStandardLibrary/blob/364a7f372da8f566e15a76f6b014ba342b75c8cb/.gitattributes#L32

it feels odd to refer from the Resources/BuildProjects to the .CI directory. Resources/BuildProjects (instead of Resources/C-Sources) might be the desired location for ModelicaUtilities.h then.

sjoelund, 2021-09-23:
Shouldn't it just be considered a system header though? It shouldn't be part of the vsproj files, right?

beutlich, 2021-09-23:
Still VS prj files need to know about such additional include directories. Same for Resources/BuildProjects/gcc.

sjoelund, 2021-09-23:
```
Resources/BuildProjects/gcc$ rm ../../C-Sources/ModelicaUtilities.h 
Resources/BuildProjects/gcc$ make CPPFLAGS=-I/home/martin/OpenModelica/build/include/omc/c 
rm -f ModelicaStandardTables.o ModelicaStandardTablesUsertab.o ModelicaMatIO.o snprintf.o ModelicaIO.o adler32.o compress.o crc32.o deflate.o gzclose.o gzlib.o gzread.o gzwrite.o infback.o inffast.o inflate.o inftrees.o trees.o uncompr.o zutil.o
rm -f *.a
rm -f ../../Library/linux64/*.a
cc -I/home/martin/OpenModelica/build/include/omc/c -DTABLE_SHARE=1 -O3 -Wno-attributes -I"../../C-Sources/zlib" -c -o ModelicaStandardTables.o ../../C-Sources/ModelicaStandardTables.c
```

PR #3871 (2021-11-16): Remove ModelicaUtilities.h from MSL distribution
ModelicaUtilities.h is ought to be distributed by Modelica tools and no longer by the MSL.

Closes #3867.
closed 2021-11-16 by MartinOtter