modelica/ModelicaStandardLibrary #3873: Quasi-static AC machine models missing under MSL Electrical

svarig, 2021-09-28:
MSL 4.0.0 distributed with Dymola 2022 (and possibly other tools) does not seem to have some packages expected for Quasi-Static AC machine models expected under Modelica.Electrical.QuasiStatic.Machines

This documentation page shows the problem

https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_Electrical_QuasiStatic_Machines.html#Modelica.Electrical.QuasiStatic.Machines

The BasicMachines package here is supposed to contain machine models but it does not.

Attached screenshot shows a view of the same MSL 4.0.0 parent package in Dymola with this issue
[image]
.

HansOlsson, 2021-09-28:
As far as I can tell this is not an issue with Dymola's distribution of the library, but with the library itself.

Compare the contents of package.mo in
https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Electrical/QuasiStatic/Machines/BasicMachines
with the directories at the same level.

HansOlsson, 2021-09-28:
It may be that the list is just copy-pasted from somewhere, if so that's an indication that copy-pasted contents-lists are problematic and not that useful.

svarig, 2021-09-28:
The quasi-static AC machine models missing here are still relevant and have use cases, If these models are still available in older MSL versions, would like to request consideration for adding them, testing & re-distributing them in future MSL versions. If they are already available in MSL under some other location, this may just need a documentation update.

AHaumer, 2021-09-28:
Sorry, there never were quastistatic AC machines, neither in Electrical.Machines nor in Electrical.QuasiStatic.Machines. 
They were planned but never implemented. Use Magnetic.QuasiStatic.FundamentalWave instead. All machine models in Magnetic are parameter and connector compatible to those in Electrical.

dietmarw, 2021-09-28:
The documentaion still needs to get updated to be in sync with the library.

beutlich, 2021-09-28:
> They were planned but never implemented.

See bddf24914fff755ceb7deabf5d3c248dc502c1a9 where it was added.

Both description and documentation of Modelica.Electrical.QuasiStatic.Machines.BasicMachines.Components also is misleading (when it was copied from Modelica.Electrical.Machines.BasicMachines.Components) and should be corrected.

closed 2021-09-30 by beutlich