modelica/ModelicaStandardLibrary #3875: Nearly-singular initialization problem in MultiBody example case

casella, 2021-09-29:
When I simulate `Modelica.Mechanics.MultiBody.Examples.Constraints.PrismaticConstraint` with Dymola, two linear systems with four equations are found in the initialization problem. Dymola solves them by tearing, and apparently the numerical solution makes physical sense. However, when inspecting the dsmodel.mof file, I found the following details:
```
  // Linear system of equations
  // Tag: initialization.linear[1]
    // Symbolic solution
      /*  Original equation
      (-5.551115123125783E-17)*freeMotionScalarInit.derv[1].u = (
        -0.036476928899629085)-(0.5318108111446445*freeMotionScalarInit.r_rel_a_2
        -0.8191520442889918*freeMotionScalarInit.r_rel_a_3+0.12158976299876358*(
        0.23702889650558212*freeMotionScalarInit.r_rel_a_2-0.0996005029250512*
        freeMotionScalarInit.r_rel_a_3-0.3)-0.6895698123465139*(0.8130157214783864
        *freeMotionScalarInit.r_rel_a_2+0.5648625214636234*freeMotionScalarInit.r_rel_a_3));
      */
      freeMotionScalarInit.derv[1].u := (-18014398509481984.0)*((
        -0.036476928899629085)-(0.5318108111446445*freeMotionScalarInit.r_rel_a_2
        -0.8191520442889918*freeMotionScalarInit.r_rel_a_3+0.12158976299876358*(
        0.23702889650558212*freeMotionScalarInit.r_rel_a_2-0.0996005029250512*
        freeMotionScalarInit.r_rel_a_3-0.3)-0.6895698123465139*(0.8130157214783864*
      freeMotionScalarInit.r_rel_a_2+0.5648625214636234*freeMotionScalarInit.r_rel_a_3)));

    // Torn part
      constraint.frame_b.r_0[3] := 0.2+0.14119408087712543*(7.082449871749603*(
        0.8130157214783864*freeMotionScalarInit.r_rel_a_2+0.5648625214636234*
        freeMotionScalarInit.r_rel_a_3)-freeMotionScalarInit.derv[1].u);
      constraint.frame_b.r_0[2] := 0.9663834860128886*freeMotionScalarInit.derv[1].u
        +0.23702889650558212*freeMotionScalarInit.r_rel_a_2-0.0996005029250512*
        freeMotionScalarInit.r_rel_a_3-0.3;
      constraint.frame_b.r_0[1] := (-1.220774588761456)*(0.5648625214636234*(
        constraint.frame_b.r_0[3]-0.2)-0.0996005029250512*constraint.frame_b.r_0[2]
        -0.19371055973531373);
  // End of linear system of equations

  // Linear system of equations
  // Tag: initialization.linear[2]
    // Symbolic solution
      /*  Original equation
      3.3306690738754696E-16*bodyOfConstraint.v_0[3] =  -(0.5318108111446445*
        freeMotionScalarInit.v_rel_a_2-0.8191520442889918*freeMotionScalarInit.v_rel_a_3
        +0.12158976299876358*(0.23702889650558212*freeMotionScalarInit.v_rel_a_2
        -0.0996005029250512*freeMotionScalarInit.v_rel_a_3)-0.6895698123465143*(
        0.8130157214783864*freeMotionScalarInit.v_rel_a_2+0.5648625214636234*
        freeMotionScalarInit.v_rel_a_3));
      */
      bodyOfConstraint.v_0[3] := (-3002399751580330.5)*(0.5318108111446445*
        freeMotionScalarInit.v_rel_a_2-0.8191520442889918*freeMotionScalarInit.v_rel_a_3
        +0.12158976299876358*(0.23702889650558212*freeMotionScalarInit.v_rel_a_2
        -0.0996005029250512*freeMotionScalarInit.v_rel_a_3)-0.6895698123465143*(
        0.8130157214783864*freeMotionScalarInit.v_rel_a_2+0.5648625214636234*
        freeMotionScalarInit.v_rel_a_3));

    // Torn part
      der(freeMotionScalarInit.derv[1].u) := 7.082449871749603*(0.8130157214783864
        *freeMotionScalarInit.v_rel_a_2+0.5648625214636234*freeMotionScalarInit.v_rel_a_3
        -bodyOfConstraint.v_0[3]);
      bodyOfConstraint.v_0[2] := 0.9663834860128886*der(freeMotionScalarInit.derv[1].u)
        +0.23702889650558212*freeMotionScalarInit.v_rel_a_2-0.0996005029250512*
        freeMotionScalarInit.v_rel_a_3;
      der(constraint.frame_b.r_0[1]) := (-1.220774588761456)*(0.5648625214636234
        *bodyOfConstraint.v_0[3]-0.0996005029250512*bodyOfConstraint.v_0[2]);
  // End of linear system of equations
```
If you look at the first of the two systems, the unknown of the residual equation is multiplied by a very small number,  -5.551115123125783E-17, which means that the symbolic solution starts with
```
freeMotionScalarInit.derv[1].u := (-18014398509481984.0)*(...)
``` 
This looks very odd to me, and apparently points to a singular initialization problem. Ditto for the other linear system.

When solving this model with OpenModelica we end up in a similar situation, with the second of the two systems having a determinant of 7.15e-17. Also the solvers used by OpenModelica manage to find a solution. However, we're now exploring algorithms to remove redundant initial equations, and in this case the algorithm detects the singular, but apparently consistent, system of equations and removes one equation from it. Then, it needs to fix an additional variable to its start value.

We agreed long time ago that the examples in the MSL should have fully specified initial conditions, to avoid the risk that different tools add different additional initial equations, leading to different results. I guess "fully specified" also mean "non-singular".

@tobolar, @MartinOtter, could you please investigate why this singularity takes place in the initialization problem, and fix it to make it non-singular, if possible?

Thanks!

henrikt-ma, 2021-09-29:
For the record, also System Modeler is not happy at all with the conditioning of this initialization problem, but also somehow manages to produce the expected result.

casella, 2021-09-29:
Good to hear 😄 

At this point we really need the expert modeller's insight on the root cause of the issue.

beutlich, 2021-09-29:
@casella Did you check the MSL master branch or outdated v4.0.0 tag?

Related to #3706?

henrikt-ma, 2021-09-30:
> @casella Did you check the MSL master branch or outdated v4.0.0 tag?

For System Modeler, this is indeed fixed in _master_: The crazy numbers involved in solving the linear systems torn down to a single variable are gone, and the numeric conditioning of the other equation systems (which I'm afraid differ a bit between tools) look great.

tobolar, 2021-09-30:
I've checked this with MSL master version and Dymola 2022 but can't find such initial equation in dsmodel.mof.

casella, 2021-09-30:
@beutlich thanks for pointing me to #3706 and #3710, I missed them. I was spoiled by you setting all the labels, sorry about that :)

We'll check with MSL master and report. I guess it would be nice to push that change also to the 3.2.3 and 4.0.0 maintenance branches, so we can use that for testing. Shall I take care of that or can you do it?

beutlich, 2021-10-01:
> We'll check with MSL master and report. I guess it would be nice to push that change also to the 3.2.3 and 4.0.0 maintenance branches, so we can use that for testing. Shall I take care of that or can you do it?

Since #3710/#3718 is labeled as bug and merged in master, I already merged it to the maint/3.2.3 (#3741) and maint/4.0.x (#3740) branch several month ago.

casella, 2021-10-01:
Aha, sorry that I didn't check. That means we are not using the HEAD versions of the maintenance branch for testing as I thought. I'll investigate why.

Thanks for the help!

closed 2021-10-01 by casella