modelica/ModelicaStandardLibrary #3878: Unit-checking using Gain and LimIntegrator-blocks

HansOlsson, 2021-10-11:
Currently Dymola as default sort of ignores unit="1"; and thus the change in https://github.com/modelica/ModelicaStandardLibrary/issues/227 is not yet active.

Trying to activate that check triggers warnings in several models, some can be traced to  `Modelica.Electrical.Batteries.BaseClasses.BaseCellStack` ( https://github.com/modelica/ModelicaStandardLibrary/blob/3057a7665290da6111c62c04ec40e29c9c8c069b/Modelica/Electrical/Batteries/BaseClasses/BaseCellStack.mo and similarly BaseCellStackWithSensors in the same package) where
```
Blocks.Continuous.LimIntegrator limIntegrator(
    final k=1/(Np*cellData.Qnom),
   ...
...
Modelica.Blocks.Math.Gain gainV(final k=Ns*cellData.OCVmax)
```
and `cellData.Qnom.unit="A.s"`, and  `cellData.OCVmax.unit="V"`.

I assume we want the improved unit-checking, and that in most cases `unit="1"` is correct, so we should change these uses - and the simplest correction in that case is to add a modifier with `unit=""` and just let normal unit-deduction work out the result.

That gives `k(unit="")=1/(Np*cellData.Qnom)`, and we can provide that as a recommendation for users.
The other alternative is `k(unit="s-1.A-1")=1/(Np*cellData.Qnom)`, but that just puts the burden on the user of the library and doesn't add anything meaningful.

I can provide a PR for it.

Also in Modelica.Electrical.Machines.Examples.DCMachines.DCPM_CurrentControlled

beutlich, 2021-10-11:
`unit=""` seems like a new concept to remove unit modifiers in favour of automatic unit-deduction. It currently is not present in MSL (master).

I am rather in favour to fix the actual unit of the gain. For example, in Modelica.Electrical.Batteries.BaseClasses.BaseCellStack.limIntegrator it is `u.unit = "A"` and `y.unit = ""`. Thus, `k.unit` should be  "A-1"? Maybe, the unit of SoC in Batteries should be changed to "1" as well.

henrikt-ma, 2021-10-12:
I find this an interesting discussion about a largely unspecified topic.  One question I find particularly interesting is whether a model with a unit error involving the inferred unit of a variable should be considered an invalid model.  That is, it would be great if we could agree that the specification should define this as a model error:
```
  Real a(unit = "m") = 1.0;
  Real b = a;
  Real c(unit = "kg") = b;
```

To clarify, the alternative interpretation that I would like to be ruled out is that the original empty unit of `b` is what should be considered in the declaration equation for `c`.

christiankral, 2021-10-13:
I need some clarification here, please.

> `unit=""` seems like a new concept to remove unit modifiers in favour of automatic unit-deduction. It currently is not present in MSL (master).

I have to admit I was not aware of the fact (so far) that using `unit=""` is a meaningful and desired option in Modelica. If I understand the discussion correctly 

- `unit=""` indicates that the unit shall be deduced automatically by the tool
- `unit="1"` indicates that the unit is equal 1 as in `R2 = k * R1` with `k(unit="")`

I assume this is somewhere written in the specification (which I could not yet find) and consequently the different meaning and intention of `unit=""` and `unit="1"` is obviously to all tool vendors. Right?

When looking at the Modelica Language Specification 3.5 I found:

[image]

So obviously, `unit = ""` is the default unit of a Real type variable. Why is it then required to use the modifier `unit=""`?

HansOlsson, 2021-10-13:
> So obviously, `unit = ""` is the default unit of a Real type variable. Why is it then required to use the modifier `unit=""`?

Because internally the Gain, Integrator and some other blocks have the modifier unit="1" due to #227, so adding unit="" for specific uses of them gets back to the original unit="", and the logical conclusion is that it should behave the same as the original (MCP-0009 might give another way of doing that).

As noted above one alternative fix for this would be to revert #227 and don't set unit for those blocks - I could make such a PR if that is desired. (There's also the possibility of mixing; but that's probably too messy.)

closed 2021-11-16 by MartinOtter

ceraolo, 2024-07-30:
I see that the topic of unit deduction and the presence of `unit="1"` is `Gain` is highly debated.
I know that there are advantages and disadvantaged of having this specification in `Gain`: 
- if it is dropped, output unit deduction is not done, at least in Dymola
- if it is there, output unit deduction is not done in case the Gain's constant has some dimension, e.g. `"s"`.

However, I see in MSL an issue of consistency. Consider as an example block `Add`. Its constants `k1` ad `k2` do not have unit `"1"`, and therefore its output unit is not possible to deduce, at least in Dymola.

This is not consistent. _(BTW, I think that is is more common to have unitless `k2` and `k2` of `Add`, than `k` of `Gain`, since `Add` is naturally used to add or substract signals, while control gains often have dimensions.)_

So, if Dymola's present behaviour is compliant to MS, the situation is inconsistent and, I would say, displaced towards the wrong direction.  But discussing whether `unit="1"` is better put on `Gain` or `Add` is subject to personal opinions.

For all this, I think that it would be very useful to drop `unit="1"` in `Gain` of MSL. In that case (with the current Dymola behaviour):
- we would lose automatic equality of `Gain` output unit to its input, but we would gain the possibility do define gains with  `k` with dimension, with automatic output unit deduction, which would be a boon
- we would gain consistency with blocks such `Add` (with all the the blocks I would say, but this should be checked).

Naturally, if this proposal is considered in principle acceptable, the whole list of blocks should be verified and made consistent with it.

casella, 2024-07-31:
See my [last comment](https://github.com/modelica/ModelicaStandardLibrary/issues/4439#issuecomment-2260555126) in #4439. I also think we should not have unit = "1" in Gain, and instead have it in Add.

Units are complicated...

PR #3881 (2021-11-16): Revert major part of #227, and remove unit from gain parameter of Modelica.Blocks.Continuous.* blocks and Modelica.Blocks.Math.Gain
Ccompare with af01213debfd82ab3e16d76346673466b4c6925b
If you want unit="1" for a gain-block, add parameter with unit="1" as in the updated test-case.
Closes #3878
commit 12900583 (2021-11-16): Revert major part of #227 - compare with af01213debfd82ab3e16d76346673466b4c6925b
If you want unit="1" for a gain-block, add parameter with unit="1" as in the updated test-case.
Closes #3878