modelica/ModelicaStandardLibrary #3887: Inadequate handling of geometric corner case in ThreePhaseTwoLevel_PWM

henrikt-ma, 2021-11-01:
During simulation of `Modelica.Electrical.PowerConverters.Examples.DCAC.MultiPhaseTwoLevel.ThreePhaseTwoLevel_PWM` I encounter the following problem under very specific conditions, leading to `pwm.svPWM.ka` = 6, which then causes array indexing out of bounds.

The problem happens when the `when`-statement in `Modelica.Electrical.PowerConverters.DCAC.Control.SVPWM` is triggered and one has (happening very close to time 0.5):
- `sine.y`: -3.4638242249419736e-16
- `pwm.u[1]`: 1.4142135623730951

Based on these numbers I then obtain:
- `pwm.svPWM.phiRef`: -2.4492935982947064e-16
- `pwm.svPWM.phiPos`: 6.2831853071795862

The `pwm.svPWM.phiPos` is equal to `2 * pi` since `pwm.svPWM.phiRef` is too small to make a difference.

For `m` = 3 we have:
- `pi / m`: 1.0471975511965976
- `pwm.svPWM.phiPos / (pi / m)`: 6 (formatted with `%.17g`)

After all, 6 should be the expected result once you have landed in `pwm.svPWM.phiPos` = `2 * pi`, and it should be clear that the model needs to be adjusted somehow to not end up in this geometric corner case.

beutlich, 2021-11-01:
You checked with current master, I believe, and as such the issues is independent of the fix #3626 for Modelica.Electrical.PowerConverters.DCAC.Control.SVPWM. Please confirm @henrikt-ma .

henrikt-ma, 2021-11-02:
Good points.  I wasn't aware of the _Correct a bug…_, but yes, the two problems are independent.  The problematic code is still present on _master_ (https://github.com/modelica/ModelicaStandardLibrary/blob/af0913b32484283822854aef4a06d98c88ac26d6/Modelica/Electrical/PowerConverters/DCAC/Control/SVPWM.mo#L40):
```
    phiRef:=if noEvent(uRef < small) then 0 else atan2(u[2], u[1]);
    //Shift angle to [0, 2*pi)
    phiPos:=max(phiRef + (if phiRef < -eps then 2*pi else 0), 0);
    //Determine sector and neighbour sector
    ka:=integer(phiPos/(pi/m));
```

Note that your mileage may vary with the quality of your tool's `ModelicaServices`, and possibly also with timing around the problematic corner case.

beutlich, 2021-11-02:
I tried to create a minimal example, but cannot reproduce in Dymola

```mo
model Issue3887
  Modelica.Electrical.PowerConverters.DCAC.Control.SVPWM svPWM(f=100, uMax=sqrt(6))
    annotation(Placement(transformation(extent={{-20,20},{0,40}})));
  Modelica.Blocks.Sources.RealExpression u1(y=1.4142135623730951)
    annotation(Placement(transformation(extent={{-52,20},{-32,40}})));
  Modelica.Blocks.Sources.RealExpression u2(y=-3.4638242249419736e-16)
    annotation(Placement(transformation(extent={{-52,2},{-32,22}})));
equation 
  connect(u1.y, svPWM.u[1]) annotation(Line(points={{-22,29.5},{-26,29.5},{-26,30},{-31,30}}, color={0,0,127}));
  connect(u2.y, svPWM.u[2]) annotation(Line(points={{-31,12},{-24,12},{-24,24},{-22,24},{-22,30.5}}, color={0,0,127}));
  annotation(uses(Modelica(version="4.0.0")));
end Issue3887;
```

Can you shre your values for Modelica.Constants.small and  Modelica.Constants.eps.

henrikt-ma, 2021-11-02:
- `eps`: 1.1102230246251565e-16
- `small`: 2.2250738585072014e-308

henrikt-ma, 2021-11-02:
Your minimal example is good (I just lowered the frequency to make debug prints more succinct).  My problem was that I accidentally used an inaccurate value for `pi`.

henrikt-ma, 2021-11-04:
To fix this, I believe it should be sufficient to just multiply `eps` by a modest factor. Without working out exact details, let's consider the current case of `m` = 3.

When adding `2 * pi` to a small negative angle, this needs to become sufficiently smaller than `2 * pi` for division by `pi / 3` to become strictly smaller than 6.

Say we want it to become at most `6 * (1 - 2 * eps)`. Since `pi / 3` is roughly 1, we get that the small angle subtracted from `2 * pi` should be roughly smaller than `12 * eps`.

Omitted details include:
- Was `2 * eps` the "right" margin of error?
- What does the "roughly" mean for the number 12?

HansOlsson, 2021-11-08:
To me the entire code seems odd:

- It is inconsistently using `noEvent `inside a when-body. That is not needed, and it is only done for some of the possibly event-generating operators.
- There should be no need to compare with -eps; just compare with 0 instead.
- The odd max-eps line could be replaced by `phiPos:=phiRef+(if phiRef<0 then 2*pi else 0)`; no `max` - no `eps`.
- Instead of trying to adjust `eps` it seems simpler to just add `if ka>=2*m then ka:=2*m-1; end if;` (or 6 and 5); and `kb:=rem(ka+1,2*m);`.

HansOlsson, 2021-11-09:
Oh, I now see why the eps was added. The example (and likely many use-cases) balance on the edge, as they generate samples 7.2 degrees apart, and thus some phiRef may be very close to zero and negative, so that there's a risk phiPos will switch from a bit negative to 360 degrees, as any minor difference is lost in roundoff.

The other logic still ensures that regardless of that we do not get values of out bounds.
That's why #2056 is important.

PR #3892 (2021-11-18): Avoid values out of bound, and simplify logic in PowerConverters.DCAC.Control.SVPWM
Closes #3887
commit 865ce07e (2021-11-18): Avoid values out of bound, and simplify logic.
Closes #3887
closed 2021-11-18 by MartinOtter