modelica/ModelicaStandardLibrary #3894: Derivative signal of combiTable2Ds is set to zero outside outside of table with HoldLastPoint

GarronFish, 2021-11-11:
The derivative signal of a Modelica.Blocks.Tables.CombiTable2Ds with extrapolation set to HoldLastPoint is zero if extrapolation is required to calculate the output.

This is incorrect for example if the input signals to a combiTable2Ds component is set to
u1=time+1
u2 = -1 
is feed into a combiTable2Ds component with 
table = [0,1,2;
             1,2,3;
             2,3,4]
and is simulated for 1[s]

The derivative of the output is zero however the output of the signal changes from 2 to 3 and the derivative should be 1.

Steps to recreate the issue in Dymola:
1. clearFlags()
2. open the tabletest3.mo (the extension will have to be renamed to .mo) and simulate by pressing F10, this should simulate and generate the plots.
3. Compare der1.y and derivative.y

der1.y is the der() of the table and this is zero when run here, derivative.y is the estimate of the derivative and converges to 1.  der1.y should be 1
[tabletest3.txt](https://github.com/modelica/ModelicaStandardLibrary/files/7519068/tabletest3.txt)
.

HansOlsson, 2021-11-11:
Is this same issue as #3893 ?

GarronFish, 2021-11-11:
It may be a different issue as this is specifically related to using extrapolation=HoldLastPoint, as this issue is not present with extrapolation=LastTwoPoints.

beutlich, 2021-11-12:
It is independent of #3893 indeed. I will fix this issue soon and review #3893 afterwards.

PR #3896 (2021-12-29): Fix derivatives in 2D-table for one-sided extrapolation by constant continuation
Closes #3894.

@GarronFish You can test by cloning master branch of [ModelicaTableAdditions](https://github.com/tbeu/ModelicaTableAdditions) where I already updated the binaries.
closed 2021-12-29 by beutlich