modelica/ModelicaStandardLibrary #3900: CombiTimeTable does not have the ability to find max/min ordinate (y) value

electricitybender, 2021-11-24:
Tables in `ModelicaStandardTables.c` (in particular the source block `CombiTimeTable`) can find the maximum and minimum abscissa (x-value, or 'time'), but cannot find the maximum and minimum ordinate (y-value, or 'data'). I would like to ask if there is any limitation or consideration keeping such a functionality from being implemented; there can be situations where knowing the min/max values inside an input table is useful before the timestepping has started.

I managed to make a copy of `Modelica.Blocks.Sources.CombiTimeTable`, add another parameter inside and call an external C function `maximimValue` as shown below.

external_C_code_additon.c:
```
/* redefine calls locally */
#define TABLE_ROW0(j) table[j]
#define TABLE_COL0(i) table[(i)*nCol]

/* make some local definitions just as in the original, for the function to work */
typedef size_t Interval[2];

typedef struct CombiTimeTable {
    char* key; /* Key consisting of concatenated names of file and table */
    double* table; /* Table values */
    size_t nRow; /* Number of rows of table */
} CombiTimeTable;

/* function */
double maximumValue(void* _tableID) {
    double yMax = 0.;
    CombiTimeTable* tableID = (CombiTimeTable*)_tableID;
    if (NULL != tableID && NULL != tableID->table) {
        const double* table = tableID->table;
        const size_t nRow = tableID->nRow;
        yMax = TABLE_ROW0(0);
        int i;
        for (i = 1; i < nRow; i++) {
        	if (TABLE_ROW0(i) > yMax) {
        		yMax = TABLE_ROW0(i);
        	}
        }
    }
    return yMax;
}
```

This will probably work for my project. I wasn't able to edit the `ModelicaStandardTables` .c and .h files as it appears they are precompiled in OpenModelica - no changes seemed to be reflected in my test models. 
Unless there are objections, I may fork this repo and produce a pull request. (I am not familiar with development for MSL so it might take a while to be able to test the functionality properly). This might be same for all other MSL table blocks but I have not explored them.

HansOlsson, 2021-11-24:
If you use spline-interpolation it may be more complicated as the maximum might be between the data-points (at least for some splines, don't know if it is relevant in MSL).

Why do you need this functionality?

electricitybender, 2021-11-24:
I'm looking at figuring out the maximum value in the input table (either a defined peak or a steady-state) as some parameters depend on it. Admittedly I don't need spline interpolation and I see why a min/max would be problematic.

beutlich, 2024-01-14:
Extrapolation is another issue to consider/disable then.

Going to close this issue as wontfix.

closed 2024-01-14 by beutlich