modelica/ModelicaStandardLibrary #3904: DC PowerSupply missing

AHaumer, 2021-11-26:
To model experimt setups from the lab, a model of a DC power supply is missing.
Normally, these devices work only in one quadrant: v > 0 and i > 0 (i out of the device).
You may set the voltage limit (constant voltage) and the current limit (constant current).
Sometimes the device ha a power limit (constant power).
I could provide a PR with such a model. Should it be placed in Electrical.Analog.Sources? What's your opinion?
Especially asking @dietmarw and @christiankral , but every feedback is welcome!

AHaumer, 2021-11-27:
I'm curious whther other simulators have problems or not.
[PowerSupply.zip](https://github.com/modelica/ModelicaStandardLibrary/files/7611441/PowerSupply.zip)

beutlich, 2022-06-03:
Resolved by #3926.

closed 2022-06-03 by beutlich