modelica/ModelicaStandardLibrary #3908: Late inlining of Modelica.Mechanics.MultiBody.Frames.Internal.maxWithoutEvent

christoff-buerger, 2021-11-30:
The current implementations of `maxWithoutEvent`, `maxWithoutEvent_d` and `maxWithoutEvent_dd` in `Modelica.Mechanics.MultiBody.Frames.Internal` all have the annotation `Inline=false`. This is not good. They should have `lateInline = true` annotated instead, to still enable inlining for performance optimization; after all, their implementations are just expressions.

@MartinOtter: Can you confirm this?

MABackoffice, 2023-02-01:
Discussion during phone meeting with @MartinOtter and Modelon India:
Change should go to 4.1.0 (not 4.0.1), as it might change behavior.

TManikantan, 2023-03-08:
@MartinOtter Hi would you please look at this issue, Please also let us know if we could help you?

TManikantan, 2023-05-12:
> m this?

@MartinOtter do you agree with this solution

TManikantan, 2023-05-12:
> The current implementations of `maxWithoutEvent`, `maxWithoutEvent_d` and `maxWithoutEvent_dd` in `Modelica.Mechanics.MultiBody.Frames.Internal` all have the annotation `Inline=false`. This is not good. They should have `lateInline = true` annotated instead, to still enable inlining for performance optimization; after all, their implementations are just expressions.
> 
> @MartinOtter, @tobolar : Can you confirm this?

@MartinOtter do you agree with this solution?

MartinOtter, 2023-05-16:
Yes, good suggestion: Replace Inline=false by lateInline=true.
This should be backwards compatible.

PR #4133 (2023-05-26): Replace 'Inline=false' with late inlining
Close  #3908
closed 2023-05-26 by MartinOtter