modelica/ModelicaStandardLibrary #3911: Severe bug in ModelicaStandardTables

AHaumer, 2021-12-07:
Taking MSL from the master branch, compiling a model with a CombiTimeTable on Windows 10 64 bit, I get the message:
Compiler message:

Compiling and linking the model (Visual C++). 

dsmodel.c
dsmodel.c(213): error C2065: 'ModelicaStandardTables_CombiTimeTable_init3': undeclared identifier
dsmodel.c(213): error C2064: term does not evaluate to a function taking 337 arguments

sjoelund, 2021-12-07:
`ModelicaStandardTables_CombiTimeTable_init3` was added quite recently. https://github.com/modelica/ModelicaStandardLibrary/pull/3691 should be for v4.0.1 which I assume Dymola doesn't include the sources for? Or something?

HansOlsson, 2021-12-07:
> `ModelicaStandardTables_CombiTimeTable_init3` was added quite recently. #3691 should be for v4.0.1 which I assume Dymola doesn't include the sources for?

It's more the order of includes and that we include the sources for 4.0.0 in the default directory.

How to handle this case is described in the Dymola manual in section 17.1.5 Additional setup>Working with a Modelica version that is newer than the one in the distribution.

beutlich, 2022-01-11:
> Taking MSL from the master branch

Note that the binaries are only updated for (tagged) releases (at least I did so in the past).

closed 2022-01-11 by beutlich