modelica/ModelicaStandardLibrary #3931: Vector/arrow visualization

beutlich, 2022-01-16:
Even after reading #2501 (where the arrow visualization was refactored to utilize the newly introduced [Vector](https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_Mechanics_MultiBody_Visualizers_Advanced.html#Modelica.Mechanics.MultiBody.Visualizers.Advanced.Vector) visualizer) I cannot comprehend why the diameter visualization was removed.

For the following example model the arrow visualization of both the SignalArrow and AbsoluteSensor works well with MSL v3.2.3. After migration to MSL v4.0.0 the arrow visualization is missing (when tested in Dymola and SystemModeler). Also the arrow diameter cannot be set.

```mo
model Unnamed
  inner Modelica.Mechanics.MultiBody.World world(animateWorld=false, animateGravity=false, label2="z", label1="x", n={0,0,-1})
    annotation(Placement(transformation(origin={-110, 30}, extent={{-10, -10}, {10, 10}})));
  Modelica.Mechanics.MultiBody.Joints.Revolute revolute1(n={0, 1, 0})
    annotation(Placement(transformation(origin={-50, 30}, extent={{-10, -10}, {10, 10}})));
  Modelica.Mechanics.MultiBody.Parts.FixedTranslation fixedTranslation1(r={0, -1, 0}, animation=false)
    annotation(Placement(transformation(origin={10, 72.396}, extent={{-10, -10}, {10, 10}})));
  Modelica.Mechanics.MultiBody.Parts.PointMass pointMass1(m=1)
    annotation(Placement(transformation(origin={52.304, 30}, extent={{-10, -10}, {10, 10}})));
  Modelica.Mechanics.MultiBody.Visualizers.SignalArrow signalArrow1(diameter=0.008)
    annotation(Placement(transformation(origin={85, 55}, extent={{-10, -10}, {10, 10}})));
  Modelica.Mechanics.MultiBody.Parts.FixedTranslation fixedTranslation2(r={1, 0, 0}, animation=false)
    annotation(Placement(transformation(origin={17.792, 30}, extent={{-10, -10}, {10, 10}})));
  Modelica.Mechanics.MultiBody.Sensors.AbsoluteSensor absoluteSensor1(get_r=true, resolveInFrame=Modelica.Mechanics.MultiBody.Types.ResolveInFrameA.world, arrowDiameter=0.008, arrowColor={255, 0, 255}) 
    annotation(Placement(transformation(origin={45, -21.932}, extent={{-10, -10}, {10, 10}})));
equation 
  connect(revolute1.frame_a, world.frame_b) annotation(Line(origin={-80, 30}, points={{20, 0}, {-20, -0}}));
  connect(revolute1.frame_b, fixedTranslation2.frame_a) annotation(Line(origin={-16.104, 30}, points={{-23.896, 0}, {23.896, 0}}));
  connect(fixedTranslation1.frame_a, world.frame_b) annotation(Line(origin={-27.5, 51.198}, points={{27.5, 21.198}, {22.5, 21.198}, {22.5, -21.198}, {-72.5, -21.198}}));
  connect(absoluteSensor1.r, signalArrow1.r_head) annotation(Line(origin={60, -16.449}, points={{-25, -16.483}, {-25, -21.483}, {25, -21.483}, {25, 59.449}}));
  connect(fixedTranslation1.frame_b, signalArrow1.frame_a) annotation(Line(origin={36.25, 63.698}, points={{-16.25, 8.698}, {-11.25, 8.698}, {-11.25, -8.698}, {38.75, -8.698}}));
  connect(fixedTranslation2.frame_b, pointMass1.frame_a) annotation(Line(origin={40.048, 30}, points={{-12.256, 0}, {12.256, 0}}));
  connect(absoluteSensor1.frame_a, pointMass1.frame_a) annotation(Line(origin={39.922, -5.546}, points={{-4.922, -16.386},{-9.922,-16.386},{-9.922,-1.386},{12.382,-1.386},{12.382,35.546}}));
  annotation(uses(Modelica(version="3.2.3")), Diagram(coordinateSystem(extent={{-150, -90}, {150, 90}}, preserveAspectRatio=true, initialScale=0.1, grid={5, 5})), Icon(coordinateSystem(extent={{-100, -100}, {100, 100}}, preserveAspectRatio=true, initialScale=0.1, grid={5, 5}), graphics={  Rectangle(lineColor={0, 114, 195}, fillColor={255, 255, 255}, extent={{-100, -100}, {100, 100}}, radius=25), Text(textColor={64, 64, 64}, extent={{-150, 110}, {150, 150}}, textString="%name")}));
end Unnamed;
```

The issue was originally reported at https://stackoverflow.com/q/70534059/8520615.

dr-mrsthemonarch, 2022-01-17:
To add to the discussion, the arrow visualisation diameters have been removed for all sensors aswell, I noticed this issue when migrating from v3.2.3 to v4 in system modeler.

tobolar, 2022-01-19:
I was only partially involved in the development of Vector, so not really know about the reasons. I suppose the reason to omit the diameter could be to disable vectors' scaling when zooming visualization scene, i.e. to make the vector visibility independent of the selected viewing point and distance.

beutlich, 2022-06-03:
@HansOlsson Can you please have a look at this issue? 🙏

HansOlsson, 2022-11-13:
I would say that diameter and length must be linked (you could trim them - but if they are scaled too independently it will be odd), so the idea was instead that automatically create a suitable diameter, instead of manually trying to set the diameter.

tobolar, 2023-06-16:
Dealing with Vector and Arrow for some reasons recently, IMO there is indeed some discrepancy in the definition and usage of these visualizers we have to clarify.

- `Arrow`: shall be a visualization object, similar to a box or a sphere. It can be used for whatever reason in an animation scene and the user shall have full control of its visualization (i.e. incl. its diameter or length/diameter of its tip). This is not the case in the current MSL version and IMO we shall revert its definition (so before #2501).
A reasonable usage is e.g. to represent the coordinate system in the `World` component.

- `Vector`: is a representation of a 3d physical quantity (except for the relative position, see below). The visualization is controlled by a tool. This is fine in the current MSL version.

- `DoubleArrow`: is generally used to visualize a torque vector. It is a backward compatible leftover and shall be replaced by `Vector` in the future. 

- *relative position*: is a dynamically visualized object, so could be represented by `Vector` as well, except for additional scaling which makes no sense - the head and the tail are given. I have no clue how to best define this now.

So to conclude: I vote for reverting (and even enhancing if we have the time) of `Arrow` for the next MSL release.