modelica/ModelicaStandardLibrary #3934: Leftovers in Modelica.Mechanics.MultiBody.World from adding vector visualizer?

otronarp, 2022-01-17:
There seems to be some leftovers in Modelica.Mechanics.MultiBody.World after adding vector visualizers in #2501.

Shouldn't `defaultN_to_m` and `defaultNm_to_m` be removed from `Modelica.Mechanics.MultiBody.World` since it's up to the tool to scale force and torque vectors?

https://github.com/modelica/ModelicaStandardLibrary/blob/40e8cf6591694b1850a192b3767215e754cb3f41/Modelica/Mechanics/MultiBody/package.mo#L140
https://github.com/modelica/ModelicaStandardLibrary/blob/40e8cf6591694b1850a192b3767215e754cb3f41/Modelica/Mechanics/MultiBody/package.mo#L143

If tools are expected to honor the `defaultArrowDiameter` that should be documented somewhere, if not the description is misleading since it gives the impression that it's used for force/torque/sensor vectors.
https://github.com/modelica/ModelicaStandardLibrary/blob/40e8cf6591694b1850a192b3767215e754cb3f41/Modelica/Mechanics/MultiBody/package.mo#L131

HansOlsson, 2022-01-17:
> There seems to be some leftovers in Modelica.Mechanics.MultiBody.World after adding vector visualizers in #2501.
> 
> Shouldn't `defaultN_to_m` and `defaultNm_to_m` be removed from `Modelica.Mechanics.MultiBody.World` since it's up to the tool to scale force and torque vectors?

I would say keep them, since they can be useful as defaults for the tools (but overridden if needed).

But may need documentation update.

> If tools are expected to honor the `defaultArrowDiameter` that should be documented somewhere, if not the description is misleading since it gives the impression that it's used for force/torque/sensor vectors.

I haven't looked at this yet. But I agree that it should be clearer.

otronarp, 2024-01-31:
Having parameters whose values are not used anywhere and are not supposed to be used anywhere in models with the only motivation that tools might want to use them sounds like a terrible idea. 

If tools wants something specific the language has vendor annotations, if MSL want's tools to look at those parameters they should be forwarded to a relevant class in `ModelicaServices`.