modelica/ModelicaStandardLibrary #3944: Bug in the SharpEdgedOrifice for Loss factors

AlesVojacek, 2022-02-08:
The Modelica.Fluid.Fittings.SharpEdgedOrifice using Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.LossFactorData.sharpEdgedOrifice for calculation of Loss factors zeta1 (for design flow) and zeta2 (for reverse flow) seems to be wrong when it compares to Idelchik.

One way to fix it:
data.zeta1 and data.zeta2 in the algorithm section of Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.LossFactorData.sharpEdgedOrifice shall be rewritten to:

data.zeta1 := ((1 - D_rel^2) + 0.707*(1 - D_rel^2)^0.375)^2*(1/D_rel^2)^2;
data.zeta2 := (k*(1 - D_rel^2)^0.75 + (1 - D_rel^2)^2 + 2*sqrt(k*(1 -D_rel^2)^0.375) + (1 - D_rel^2))*(1/D_rel^2)^2;

wischhusen, 2022-03-01:
I double-checked the formular in Source Idelchik. The ratio D_rel is actually a ratio of cross-rectional areas. For circular cross-sections we receive using the original nomenclature F0/F1 = D0^2/D1^2.
I would suggest to exchange D_rel by A_rel and put leastDiameter/diameter to the power of 2.

   Real A_rel=(leastDiameter/diameter)^2;
   Real LD=length/leastDiameter;
   Real k=0.13 + 0.34*10^(-(3.4*LD + 88.4*LD^2.3));
algorithm 
   data.diameter_a := diameter;
   data.diameter_b := diameter;
   data.zeta1 := ((1 - A_rel) + 0.707*(1 - A_rel)^0.375)^2*(1/A_rel)^2;
   data.zeta2 := k*(1 - A_rel)^0.75 + (1 - A_rel)^2 + 2*sqrt(k*(1 -
     A_rel)^0.375) + (1 - A_rel);

wischhusen, 2022-03-01:
Please also compare:

Modelica.Fluid.Dissipation.PressureLoss.Orifice.dp_thickEdgedOverall_MFLOW() and
Modelica.Fluid.Dissipation.PressureLoss.Orifice.dp_thickEdgedOverall_DP()

PR #4258 (2024-01-16): Bug in SharpEdgedOrifice for Loss factors resolved
This follows https://github.com/modelica/ModelicaStandardLibrary/pull/3825#pullrequestreview-1375301502 and extracted the original commit of @wischhusen https://github.com/wischhusen/Modelica-1/commit/a0002e1f645324ecbff2e09080b15a359aa64ee3 to resolve #3944.
closed 2024-01-16 by beutlich