modelica/ModelicaStandardLibrary #3954: Broken link to academia.edu

beutlich, 2022-02-22:
As reported by https://github.com/modelica/ModelicaStandardLibrary/runs/5262352623?check_suite_focus=true the reference to acdemia.edu for "J. P. Holman, <em>Heat Transfer</em>, 10th edition, McGraw-Hill, 2010" is broken.

beutlich, 2022-03-27:
Also the revised link got broken again.

bilderbuchi, 2022-04-02:
I tried to find a homepage for the book on the publisher's site, but that seems to be impossible(?). The ISBN is 978-007-126769-4, maybe it makes sense to include that instead of a link of limited lifetime. I have it on my desk if you need any more info.

beutlich, 2022-04-02:
Yes, we should simply remove that link.

Btw, I downloaded this PDF when it still was available.

AHaumer, 2022-04-03:
I didn't find the book at the publisher's site, too. It seems that you can only buy it second hand. Maybie due to that fact non-permanent links to pdfs are published from time to time. Two solutions: Remove the links and replave it by mentoning the ISBN or searching for a newer text book covering that topic.

dietmarw, 2022-04-04:
Looking around I found that the "new" link is https://www.academia.edu/6340170/Heat_Transfer_J_P_Holman_Ed_10 but here you need to login and also need to be careful to not sign up for some subscription (which is not needed but needs to be declined). Since I found the link simply by googling it I'd say other normal users would be able to do the same. So the information of title, author, ISBN should be sufficient without any link.

PR #3955 (2022-02-26): Fix broken link to academia.edu
Fix #3954
PR #3981 (2022-05-13): Fix broken link to academia.edu
Fixes #3954.
closed 2022-05-13 by beutlich