modelica/ModelicaStandardLibrary #3960: Documentation image M_OLine error

paultjevdh, 2022-03-14:
Hello,
It seems the documentation image at modelica://Modelica/Resources/Images/Electrical/Analog/segment.png has some errors.
The component subscripts should indicate the lines between which the component is connected. 
In the image C41 is however connected in parallel to C42. C42 should be connected between nets n4 and n2 instead of n4 and n1.
Also C31 is connected to n2 rather than n1.
Finally G31 is connected to n2 instead of n1.
[image]

A proposal for the fixed image is below:
[image]

Best regards,
Paul van der Hulst

dietmarw, 2022-03-15:
It looks like these old versions of the images were forgotten to get removed when #3748 was fixed by #3756 and  #3815. 
The new versions can be found in `Modelica/Resources/Modelica/Resources/Images/Electrical/Analog/Lines/segment.png` and `Modelica/Resources/Modelica/Resources/Images/Electrical/Analog/Lines/segment_last.png`, they are used in the documentation of: `Modelica.Electrical.Analog.Lines.M_OLine`.

@max-privato and @AHaumer, do you agree that the old versions: 
 - `Modelica/Resources/Modelica/Resources/Images/Electrical/Analog/segment.png`
 - `Modelica/Resources/Modelica/Resources/Images/Electrical/Analog/segment_last.png`
 - `Modelica/Resources/Modelica/Resources/Images/Electrical/Analog/segment_first.png`
can be safely removed since they are not used anywhere in the MSL any more?

paultjevdh, 2022-03-15:
Hi,
You're completely correct. I missed that this was fixed in the current development code.
This leaves me with the question when 4.1 is due...

Best regards,
Paul

dietmarw, 2022-03-15:
@paultjevdh  Unfortunately there is currently no timeline available. The good new is that the current development version can be used directly in your tool given that your tool has been set up for MSL 4.0.0.

max-privato, 2022-03-16:
> It looks like these old versions of the images were forgotten to get removed when #3748 was fixed by #3756 and #3815. The new versions can be found in [...]

Agree. they are bad, never used and created portential misunderstandings.
BTW I like much more the new pictures that I mysself  created :-)

paultjevdh, 2022-03-16:
I agree, much better. Thanks!

commit a93ce277 (2022-06-15): Removes outdated segment images from OLine (#3961)

Closes #3960
closed 2022-06-15 by beutlich