modelica/ModelicaStandardLibrary #3984: Read real matrix from file not working - Modelica.Utilities.Examples.ReadRealMatrixFromFile

luizmac, 2022-05-06:
Hi,

I am trying to read a matrix from a .mat file but the function readRealMatrixFromFile can't read my .mat file. 

I have created a model based on the model Modelica.Utilities.Examples.ReadRealMatrixFromFile. This file can read the files .mat provided in the example (file Test_RealMatrix_v445.mat) but when I use a file generated from GNU Octave or MATLAB the function can't read it. I have tried with all formats available for .mat files in MATLAB (2020a) and GNU Octave (7.1). However no one could be properly read by Openmodelica, the function just read properly the .mat provided with the example.

I am using OpenModelica 1.18.1 on Linux and the compilation output is:

-------------
Simulation process failed. Exited with code 255.
/tmp/OpenModelica_luiz/OMEdit/Ler_Matrizes/Ler_Matrizes -port=35187 -logFormat=xmltcp -override=startTime=0,stopTime=1,stepSize=0.002,tolerance=1e-6,solver=dassl,outputFormat=mat,variableFilter=.* -r=/tmp/OpenModelica_luiz/OMEdit/Ler_Matrizes/Ler_Matrizes_res.mat -w -lv=LOG_STATS -inputPath=/tmp/OpenModelica_luiz/OMEdit/Ler_Matrizes -outputPath=/tmp/OpenModelica_luiz/OMEdit/Ler_Matrizes
... loading "Matrix_A" from "/usr/lib/omlibrary/Modelica 4.0.0/Resources/Data/Utilities/Test_RealMatrix_v7.mat"

... loading "Matrix_A" from "/usr/lib/omlibrary/Modelica 4.0.0/Resources/Data/Utilities/Test_RealMatrix_v6.mat"

... loading "Matrix_A" from "/usr/lib/omlibrary/Modelica 4.0.0/Resources/Data/Utilities/Test_RealMatrix_v4.mat"

... loading "Rr" from "/mnt/Dados/Estudos/Estudos/Modelica/DC-AC/LCL 1f/Matrizes2.mat"

Not possible to open file "/mnt/Dados/Estudos/Estudos/Modelica/DC-AC/LCL 1f/Matrizes2.mat": No such file or directory

simulation terminated by an assertion at initialization
-----------

On the annex I am providing my model for read the matrix and two .mat files, the file Matrizes2.mat (created on Octave using .mat -v6  version) which is not working and file Test_RealMatrix_v445.mat (this one provided with the example) which is working well

[FIles.zip](https://github.com/modelica/ModelicaStandardLibrary/files/8640177/FIles.zip)
.

beutlich, 2022-05-06:
When testing in Dymola 2022x I observed a dimension problem. Hence, this looks like a duplicate of #3608 to me which got fixed by #3609 for next MSL release.

luizmac, 2022-05-09:
Thanks! Problem solved. I implemented the suggested solution and my code worked properly.

closed 2022-05-09 by luizmac