modelica/ModelicaStandardLibrary #3991: ForceAndTorque example uses deprecated scaling

anotheruserofgithub, 2022-06-09:
`Modelica.Mechanics.MultiBody.Examples.Elementary.ForceAndTorque` uses the `World` component to convert coordinates of vectors from [N(m)/m] to [m] for scaling force/torque arrows:
https://github.com/modelica/ModelicaStandardLibrary/blob/1dd53117cde44b3b20fc35079b79f7ba34af0d5f/Modelica/Mechanics/MultiBody/Examples/Elementary/ForceAndTorque.mo#L4

But from what I can understand from #3931 & #3934, those parameters are deprecated.
Hence one would expect the tool to use the default scale factors 1000, which by the way can be found not only in `World` but also in `Types.Defaults`, commented out (why?):
https://github.com/modelica/ModelicaStandardLibrary/blob/1b91ac4227046cd3e87a7ebf840074cb16445cc6/Modelica/Mechanics/MultiBody/Types/Defaults/package.mo#L34-L39

However, doing so would produce a bad animation for the typical example `Examples.Elementary.ForceAndTorque`, as the torque arrow would be much smaller than the force arrow, and even smaller than the usual length of the two head cones.

I am currently implementing support for animation of the Vector visualizer in OpenModelica, and therefore I chose default scale factors of respectively 1200 and 120 for force and torque in order to replicate the image produced in the documentation of the model.

What is the best direction for selecting the scales? Should default values in `World` still be used? If not, how to choose them, what are the possible references?
I mean, if for instance there is only one force and one torque, should they have the same length? But when there are more than one vector for a quantity, what should be done?
It seems to me that it would have been much simpler to let the user/modeler set all the lengths themselves, instead of relying on the tools which will then probably display arrows with very different lengths for the same model, thus breaking in some way the interoperability across tools.
Nonetheless I understand that the lengths of the vectors are not meant to represent physical quantities but relative strength, and also that this was done (as far as I understood) to possibly avoid scaling vectors like other shapes when zooming in/out the window.

So what are your recommendations? Should we maintain the old behavior of `Examples.Elementary.ForceAndTorque`, i.e. scales of 1200 & 120?
If yes, should those be the default values for all force and torque vectors or should they be retrieved from the world parameters?
If no, how to proceed, should we scan all the shapes in the animation to get their lengths and compute arrow lengths so that they fit nicely in the window? Should we consider orientation of the animation as well?

Sorry for all those questions but I wonder why it should be up to the tools to make so many subjective (and complicated) choices, while if they were not supposed to take such decisions, then users could have full control over the displayed elements and always be cross-tool compatible.

Thank you for your feedback!

anotheruserofgithub, 2022-09-17:
For OpenModelica I implemented some heuristics in OpenModelica/OpenModelica#9389 for the choice of suitable radius and length scales for vector visualizers, as well as zoom-invariance and drawing-on-top features.
See also OpenModelica/OpenModelica#9390 for a proposal on using [vendor-specific annotations](https://specification.modelica.org/master/annotations.html#vendor-specific-annotations) to let the user choose more appropriate configurations and parameters.

I'm closing this ticket since it's no longer an issue.
Please feel free to share your feedback or any other idea you may have.

closed 2022-09-17 by anotheruserofgithub

HansOlsson, 2022-11-11:
(Going through my backlog...)

In Dymola we actually pick up the default-scaling from `defaultNm_to_m `and `defaultN_to_m`, but allow the user to override it without retranslating the model. 

(Well, we currently only support it for one world-object.)