modelica/ModelicaStandardLibrary #3992: Transparency input for all MultiBody visualizers

anotheruserofgithub, 2022-06-10:
### Feature Request

There are use cases where it could be great to be able to change the transparency of any visualizers in the animation window.
For instance, OMEdit allows to do that through a context menu (see openmodelica/openmodelica#9083), but this has to be done each time you simulate the model as the values are not persisted.
Instead, it would be very useful to have a parameter, or even a time-varying input, that can be set in the model directly.
[Surface](https://github.com/modelica/ModelicaStandardLibrary/blob/0d643bb57bdc262a6e0e8281b010c9021a00ac30/Modelica/Utilities/Internal.mo#L74) already has [that](https://github.com/modelica/ModelicaStandardLibrary/blob/0d643bb57bdc262a6e0e8281b010c9021a00ac30/Modelica/Utilities/Internal.mo#L101), but not [Shape](https://github.com/modelica/ModelicaStandardLibrary/blob/0d643bb57bdc262a6e0e8281b010c9021a00ac30/Modelica/Utilities/Internal.mo#L11) nor [Vector](https://github.com/modelica/ModelicaStandardLibrary/blob/0d643bb57bdc262a6e0e8281b010c9021a00ac30/Modelica/Utilities/Internal.mo#L47), while I do not see what could prevent them from having such a feature.
They all have a color, so why not a transparency?

HansOlsson, 2022-06-12:
Historically one major reason is that transparency for _multiple_ object is problematic; both in terms of what it implies (what happens if you split an object into two - will that be less transparent?) and in terms of implementation (ideally special shaders, depth peeling, OIT-variants).

Making one object transparent that sort of works is trivial in comparison.

anotheruserofgithub, 2022-06-12:
I'm new to this topic and didn't know it was so difficult. Thank you very much for the answer and the depth peeling / OIT pointers!

tobolar, 2022-06-22:
@HansOlsson If I understand correctly, the thing is the ambiguous definition of "transparency".

HansOlsson, 2022-06-22:
> @HansOlsson If I understand correctly, the thing is the ambiguous definition of "transparency".

There are two factors - it's ambiguous and it's difficult to implement in a good way.

anotheruserofgithub, 2022-06-22:
Yet, why does Surface have a transparency input, then?
Should it not be removed to be consistent across visualizers?
Sorry but I forgot to ask the question earlier.

If it is kept, it would be a slight improvement to modify the documentation: "Transparency of surface" instead of "[Transparency of shape](https://github.com/modelica/ModelicaStandardLibrary/blob/0d643bb57bdc262a6e0e8281b010c9021a00ac30/Modelica/Utilities/Internal.mo#L101)", not only in PartialSurface but in all models that make use of Surface (Rectangle, Torus, PipeWithScalarField).

anotheruserofgithub, 2022-06-25:
@HansOlsson Would you have an answer? (Sorry for asking late.)

anotheruserofgithub, 2022-06-27:
> Removing things is a difficult task as some might use it.
> 
> Especially as if it is only one Surface it should work. (Well, only one surface visible per pixel; so you could have a non-overlapping set of surfaces as well).

Understood, thanks! :)