modelica/ModelicaStandardLibrary #3996: Silent error handling for Modelica.Utilities.Files.copy

GarronFish, 2022-06-17:
The Modelica.Utilities.Files.copy command currently causes an error which halts any function that is being run.  It would be useful to have an option in Modelica.Utilities.Files.copy where the errors could be displayed and an error code (or error status and error message) returned from the copy function and the calling function determine what action to take.

sjoelund, 2022-06-17:
Do you have an example that triggers the message that is never shown? In OpenModelica I get:
```mo
model M
  Real r;
algorithm
  r := time;
  Modelica.Utilities.Files.copy("asfdlkasfjklaf", "x");
end M;
```
```
marsj@WIN01811:~/tmp$ ./M
assert            | debug   | It is not possible to copy the file or directory
|                 | |       | "asfdlkasfjklaf" because it does not exist.
assert            | info    | simulation terminated by an assertion at initialization
```

GarronFish, 2022-06-17:
`function processFiles
  input String filesToProcess[:] "Files to be processed";
protected
  String fileName "Current file being processed";
algorithm
  for fileName in filesToProcess loop
        success = Modelica.Utilities.Files.copy(fileName, "x");
        if success then
              // continue with processing x
        else
             // generate a warning message saying that fileName was not processed due to an error during copying
        end if;
   end for;
      
end processFiles;`

sjoelund, 2022-06-17:
`Files.copy` does not return success though. If it fails, it prints the error-message.

beutlich, 2024-01-13:
> `Files.copy` does not return success though. If it fails, it prints the error-message.

That's exactly what the issue is about: Modelica.Utilities.Files.copy (and probably more) should be error tolerant.

Here's my current workaround (with redundant checks and open to race conditions):

```
model M
  Real r;
  Modelica.Utilities.Types.FileType fileType;
  parameter String srcFilePath = "asfdlkasfjklaf";
algorithm
  r := time;
  fileType := Modelica.Utilities.Internal.FileSystem.stat(srcFilePath);
  if fileType == Modelica.Utilities.Types.FileType.RegularFile then
      Modelica.Utilities.Files.copy(srcFilePath, "x");
  else
      assert(fileType == Modelica.Utilities.Types.FileType.RegularFile, "\"" + srcFilePath + "\" cannot be copied", AssertionLevel.warning);
  end if;
end M;
```