modelica/ModelicaStandardLibrary #4012: Remove rotational coupling

tobolar, 2022-07-05:
I don't understand the meaning of the component `Modelica.Mechanics.Rotational.Components.Coupling` introduced with #3786. Even reading the documentation does not help.

Considering the `Coupling` icon, I would expect similar (or even the same) behaviour like `Modelica.Mechanics.Rotational.Components.Disc`, since it makes an impression of a gear shift (formschüßige Kupplung). Regarding behaviour (equations), it behaves exactly like `Modelica.Mechanics.Rotational.Components.IdealGear` with `ratio=-1`.

The example `Modelica.Electrical.Machines.Examples.DCMachines.DCPM_Drive` even works fine with `Rotational.Components.IdealGear idealGear(ratio=-1)`.

tobolar, 2022-07-18:
I wonder how the real mechanism looks like. "two electric machines at the shaft" - this is what confuse me, because one shaft could hardly rotate in different directions.
The icon reminds me of a jaw coupling, but the behaviour of the component seems be different.

AHaumer, 2022-07-18:
Jaw coupling is the right idea:
[image]
For electric machines, positive direction (right) is defined clockwise when looking at the front of the shaft end. 
Therefore one machine rotates cw (right), the coupled machine ccw (left).

beutlich, 2022-07-18:
I am with @tobolar here: If Coupling simply is a specific paramerization of IdealGear, it should be either removed again or implemented as extension (e.g. short class definition) to make this dependency transparent.

AHaumer, 2022-07-20:
Well a coupling primarily is _not_ an IdealGear, so extending (including inheriting the icon) is not a good idea. 
Could use the IdealGear inside the coupling model with its own icon.

AHaumer, 2022-07-29:
The misunderstanding stems from the following:
Looking at the system, both coupled machines turn in the same direction. 
Looking at each of the machines separately, one machine turns left (clockwise), the other one turns right (counterclockwise). 
This affects the induced voltage as well as control of the machine. 
Control of one machine doesn't take into account the other machine (the load).

tobolar, 2022-08-04:
@AHaumer I can follow your explanation.
Considering again the example `Modelica.Electrical.Machines.Examples.DCMachines.DCPM_Drive` is also helpful: when connecting `dcpm1`directly to `dcpm2`, it would result in the same rotation of both machines. 

So a solution is needed to "induce" an opposite rotation.

I see several possibilities.
1. To introduce a flag in the machine(s) to enable changing the sign between the mechanical and electrical torque/rotation. Then, the machines could be connected directly.
2. Since `Coupling` is a particular solution for a particular example, it could be moved e.g. into `Modelica.Electrical.Machines.Examples.Utilities`. As a mechanical component `Modelica.Electrical.Machines.Losses.Friction` also does.
3. To use `IdealGear` with `ratio=-1` instead of `Coupling`. The model is mechanically wrong but works as expected. Btw. this is the case of `Coupling` in the current implementation as well.
4. Rename and redesign icon (jaw coupling-like is misleading) of `Coupling` and enhance the documentation to be clear what is the component intended for.

AHaumer, 2022-08-05:
Addressed by #4019 -> closed.

tobolar, 2022-08-05:
I also tend to 4.

> Why do you dislike the icon? 

Well, the icon make an impression the component behaves like a *jaw coupling*, but it, in fact, behaves quite differently. This is simply misleading to users.

AHaumer, 2022-08-05:
Well, actually the device acts lika a jaw coupling, indeed.
It's only a matter of how cw and ccw rotation is defined for the coupled machines. 
For me it's completely clear - maybe electric machines are the only ones for which this is defined.
But I can't imagine that this is not defined clearly for other machines, e.g. pumps, fans, etc.
If a pump is only allowed to be rotated cw, you have to operate the electric machine correctly.
No I'll keep the jaw coupling icon.

casella, 2025-03-01:
@AHaumer, you closed this issue in 2022, but PR #4019 is still pending and was never merged, so this is still to be resolved.

casella, 2025-03-05:
@AHaumer, ping!. Sorry for pushing, but we really need to get the the release candidate ASAP. Thanks!

AHaumer, 2025-03-05:
@casella sorry ping! might disappear in the flood of messages just before the release.
I've updated PR #4019, waiting for review.

casella, 2025-03-08:
Once #4019 is merged in, it should still be back-ported by @Esther-Devakirubai and the reference results for ModelicaTest.Rotational.AllComponents updated by @MatthiasBSchaefer.

Esther-Devakirubai, 2025-03-10:
Backported by https://github.com/modelica/ModelicaStandardLibrary/pull/4584

casella, 2025-03-10:
@MatthiasBSchaefer this is now ready for ModelicaTest.Rotational.AllComponents reference result update.

casella, 2025-03-11:
We should check that it's mentioned in the Release Notes, then we can close this

AHaumer, 2025-03-11:
@Esther-Devakirubai please add an item in the release notes as new component:
Modelica.Electrical.Machines.Utilities.RotationalCoupling 
Ideal stiff coupling: One device connected to the coupling is turning clockwise (looking at the shaft end), the other device connected to the coupling is turning counter-clockwise (looking at the shaft end). The torque at flange_b has the same magnitude as the torque at flange_a but opposite sign.

casella, 2025-04-11:
Done in 51c21d4

closed 2025-04-11 by casella