modelica/ModelicaStandardLibrary #4026: Obsolete tool-specific hack in 'state1' and friends?

henrikt-ma, 2022-08-31:
What lies behind these function dummy inputs?
https://github.com/modelica/ModelicaStandardLibrary/blob/3a5289d617824d1b4f04ca133a065c5532b5df8d/Modelica/Blocks/Interfaces.mo#L1647

Perhaps the time is ripe for removing them (tools should have had ample time to fix the problem).  If not, I suppose the problem is so tricky to deal with that the documentation should go more into details about why they are still needed.

HansOlsson, 2022-08-31:
It was reintroduced in January 2016 (and then it has been moved and copied a number of times - tracking code through such changes is kind of messy).

https://github.com/modelica/ModelicaStandardLibrary/commit/3432a2218e1a4ff83dc1bd67f799aac6f6644eb4

As far as I could see it should work after that time, but there might be something that I missed.

HansOlsson, 2022-09-01:
Thinking more I realize that if you creatively use the function to send in a vector that doesn't satisfy the requirement the extra argument is sometimes needed - consider
  `state1({1,1} /*, time*/)`
The intended derivative is `1` - but the function call is time-independent without the dummy-argument and thus there's no reason to differentiate it.

henrikt-ma, 2022-09-01:
But why care about the consequences of invalid usage?  I mean, the function requires the second array entry to be the derivative of the first.

HansOlsson, 2022-12-08:
> But why care about the consequences of invalid usage? I mean, the function requires the second array entry to be the derivative of the first.

Ideally, but as far as I recall the invalid usage isn't that uncommon. Messing around with fake-derivatives and states is something that people do - and one counter to that is to make it easier to do it correctly, e.g., https://github.com/modelica/ModelicaSpecification/pull/3164