modelica/ModelicaStandardLibrary #4030: Bug in a component: Modelica.Electrical.Machines.Sensors.HallSensor

ankitnaik177, 2022-09-14:
The block Modelica.Electrical.Machines.Sensors.HallSensor has an error.  It uses rem() but it really should be mod(). 

The hall sensor should always be between 0 and 2*pi — no matter the direction of travel.  The electrical angle of a motor is always in that range — as soon as you reach 2 Pi you wrap back to zero and if you pass zero in the negative direction, you wrap to 2 pi. Note that if the angle is negative, the HallSensor gives a nonsensical output.

dietmarw, 2022-09-15:
@AHaumer How severe do you reckon this fault is? Does justify backporting to 4.0.x branch?

AHaumer, 2022-09-15:
To be honest: the difference between the two variants is 2*pi - so I wouldn't care for backporting.

dietmarw, 2022-09-15:
OK. In that case, I'll remove the "bug" label from #4031.

PR #4031 (2022-10-07): Fix HallSensor
 Fix #4030  - thanks @ankitnaik177 for pointing this out.
commit 01c3f90f (2022-10-07): fix #4030
closed 2022-10-07 by AHaumer