modelica/ModelicaStandardLibrary #4058: MoistAir, error in massFraction_pTphi

GaelEnee, 2022-11-15:
Hello, 

I would like to report an error in Modelica.Media.Air.MoistAir.massFraction_pTphi.

I suggest the replacement of :
`protected 
  constant Real k=0.621964713077499 "Ratio of molar masses";
  AbsolutePressure psat=saturationPressure(T) "Saturation pressure";
algorithm 
  X_steam := phi*k/(k*phi + p/psat - phi);`
by :
`protected 
  AbsolutePressure psat=saturationPressure(T) "Saturation pressure";
algorithm 
  X_steam := phi*k_mair/(p/psat - phi);`

Thanks !

GaelEnee, 2022-11-15:
Apologies,

After a double check, I would rather modify the expression in `Modelica.Media.Air.ReferenceMoistAir.massFraction_pTphi` :

`X := {phi*k_mair/(p/pds - phi),1 - phi*k_mair/(p/pds - phi)};`

by: 

'X := {k_mair/(k_mair + p/pds/phi -1),1 - k_mair/(k_mair + p/pds/phi -1)};

Sorry.

thorade, 2022-11-16:
https://github.com/modelica/ModelicaStandardLibrary/blob/c15f9d3bea31db7a03904248c81c77f26bb6cb2f/Modelica/Media/Air/MoistAir.mo#L242-L257

thorade, 2022-11-16:
@GaelEnee Could you help me understand this quicker by elaborating what is wrong and why? Is the corrected version taken from any reference?

mestinso, 2023-01-06:
Just glancing quickly at this, there appears to be at least one wrong assumption that *may* be guiding you to a wrong conclusion:  Note that ReferenceMoistAir doesn't follow ideal gas law (like the non-reference version).

GaelEnee, 2023-01-11:
Hello,
The presumed mistake I reported looks much like a scripting mistake to me. Getting rid of the ideal gas assumption for passing from the 4th to the 5th equation would involve the compressibility modulus Z of steam and air (pV=Z(p, T)nRT) and would result in a much more complicated final expression for X_steam. If you can get in touch with the original author of the function, he might provide an relevant advice about it.