modelica/ModelicaStandardLibrary #4060: Standard Water Issue

mgithubk, 2022-11-24:
I am trying to get the density of water using the standard water library.

The command Modelica.Media.Water.StandardWaterOnePhase.density_pT(101325,275,1,1) returns 999.9

however, the same command does not work in a function.

It yields the following error:

Undeclared function Modelica.Media.Water.WaterIF97_pT.density_pT_Unique1.
Undeclared function Modelica.Media.Water.WaterIF97_pT.density_pT_Unique1.
Failed to expand Modelica.Media.Water.WaterIF97_pT.density_pT_Unique1(101325, 278, 1, 1).
Failed to expand BasicFlowComponent.vaporMassTransport(275, 101325, 278, 5, 1, 0.001).

The function is provided here:

function vaporMassTransport
  //Testing BasicFlowComponent.vaporMassTransport(275,101325,278,5,0.001) Modelica.Media.Water.StandardWater.density_pT(pressAir,tempSurf)

  input Real tempAir "air temperature";
  input Real pressAir "air pressure";
  input Real tempSurf "surface temperature";
  input Real convCoeff "heat transfer coefficient";
  input Real area " discharge coefficient";
  input Real massPerVap "mass fraction";
//Modelica.Media.Water.StandardWaterOnePhase.ThermodynamicState stateDensH2O(p = 101325,T = 375);

  Modelica.Media.Air.MoistAir.ThermodynamicState stateAir(p = pressAir,T = tempAir,X = {massPerVap, 1-massPerVap});
  output Real vaporMassFlow "pressure drop";

protected 
  Real diffAirWat =  6.429e-11 * tempAir^2 - 1.048e-8*tempAir + 1.067e-6;
  Real specHeatAir = Modelica.Media.Air.MoistAir.specificHeatCapacityCp(stateAir);
  Real densAir =  Modelica.Media.Air.MoistAir.density(stateAir);
  Real condAir = Modelica.Media.Air.MoistAir.thermalConductivity(stateAir);
  Real diffThermAir = condAir / (densAir * specHeatAir);
  Real hm = (convCoeff / (densAir * specHeatAir)) * (diffAirWat/diffThermAir)^(2/3);
  // Need the vapor pressure of water in the air.
  Real pVap[1] = Modelica.Media.Air.ReferenceMoistAir.Utilities.pd_pTX(pressAir,tempAir,[massPerVap, 1-massPerVap]);
  Real pSurf = Modelica.Media.Air.MoistAir.saturationPressure(tempSurf);
  Real densH2O =   Modelica.Media.Water.StandardWaterOnePhase.density_pT(pressAir,tempSurf,1,1);

algorithm 

 vaporMassFlow := hm*densH2O*area*(pSurf - pVap[1])/pressAir;
end vaporMassTransport;

mestinso, 2023-01-03:
I don't know exactly what the root issue is, but, as a workaround, I expect you should be able to use Modelica.Media.Water.IF97_Utilities.rho_pT directly instead.

hubertus65, 2024-11-11:
To me this looks like an issue with the tool you use: please file with the tool vendor. I tried this in Modelon Impact, and it works fine without an error and the expected result.

closed 2024-11-11 by hubertus65

casella, 2024-11-11:
I tested this model in OpenModelica, latest nightly build, and it worked fine.

```modelica
model Test
function vaporMassTransport
  //Testing BasicFlowComponent.vaporMassTransport(275,101325,278,5,0.001) Modelica.Media.Water.StandardWater.density_pT(pressAir,tempSurf)

  input Real tempAir "air temperature";
  input Real pressAir "air pressure";
  input Real tempSurf "surface temperature";
  input Real convCoeff "heat transfer coefficient";
  input Real area " discharge coefficient";
  input Real massPerVap "mass fraction";
//Modelica.Media.Water.StandardWaterOnePhase.ThermodynamicState stateDensH2O(p = 101325,T = 375);
  output Real vaporMassFlow "pressure drop";
protected
  Modelica.Media.Air.MoistAir.ThermodynamicState stateAir(p = pressAir,T = tempAir,X = {massPerVap, 1-massPerVap});

protected 
  Real diffAirWat =  6.429e-11 * tempAir^2 - 1.048e-8*tempAir + 1.067e-6;
  Real specHeatAir = Modelica.Media.Air.MoistAir.specificHeatCapacityCp(stateAir);
  Real densAir =  Modelica.Media.Air.MoistAir.density(stateAir);
  Real condAir = Modelica.Media.Air.MoistAir.thermalConductivity(stateAir);
  Real diffThermAir = condAir / (densAir * specHeatAir);
  Real hm = (convCoeff / (densAir * specHeatAir)) * (diffAirWat/diffThermAir)^(2/3);
  // Need the vapor pressure of water in the air.
  Real pVap[1] = Modelica.Media.Air.ReferenceMoistAir.Utilities.pd_pTX(pressAir,tempAir,[massPerVap, 1-massPerVap]);
  Real pSurf = Modelica.Media.Air.MoistAir.saturationPressure(tempSurf);
  Real densH2O =   Modelica.Media.Water.StandardWaterOnePhase.density_pT(pressAir,tempSurf,1,1);
algorithm 
 vaporMassFlow := hm*densH2O*area*(pSurf - pVap[1])/pressAir;
end vaporMassTransport;
  Real w = vaporMassTransport(275,101325,278,5,0.001,0.001);
end Test;
```

Please note that non-input, non-output variables in function should be protected.