modelica/ModelicaStandardLibrary #4069: Problematic to compare h_outflow when m_flow is zero

henrikt-ma, 2023-01-11:
This is a question regarding the [comparison signals](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0/ModelicaTest/Fluid/TestComponents/Valves/TestValvesCompressibleReverse/comparisonSignals.txt) for `ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse`.

From time 3.0 until the end at time 4.0, `V5.port_b.m_flow` is zero.  Doesn't this make `V5.port_b.h_outflow` a bad comparison signal?

(If that's the case, it might be a good idea to go through all Fluid reference results and look for similar problems.)

HansOlsson, 2023-01-12:
I don't see that h_outflow is a bad signal in those cases.
As I understand the actual value shouldn't matter, but since it is given internally in the model it should still be possible to use it for regression testing.

henrikt-ma, 2023-01-12:
I can't see how it is given internally in the model in such a way that it would be useful for regression testing.  Could you please elaborate?

What I see is just what's in `Modelica.Fluid.Valves.BaseClasses.PartialValve`:
```
  port_a.h_outflow = inStream(port_b.h_outflow);
  port_b.h_outflow = inStream(port_a.h_outflow);
```
To me this looks like the standard machinery of `inStream` would get involved, meaning that results may be undefined when `m_flow` is zero.