modelica/ModelicaStandardLibrary #4072: Near-zero coordinates i graphical annotations

DagBruck, 2023-01-18:
Many models in MSL contain near-zero coordinates in graphical annotations, for example 

`annotation (Line(points={{5.55112e-16,-49},{0,-42}}, color={0,0,127}));`

The most likely historical cause for these coordinates is a rounding-bug in Dymola. Anticipating the planned MSL-cleanup project, I have search for likely candidates that should be replaced by `0`, see attached list: [nearzero.txt](https://github.com/modelica/ModelicaStandardLibrary/files/10445002/nearzero.txt)

dietmarw, 2023-01-23:
@DagBruck Does Dymola also have the ability to correct those near-zero coordinates? If yes then it would be an easy fix and you could provide a PR for this.

beutlich, 2023-01-26:
I previously cleaned them up manually, see for example 9b1f1437f5c7faa71b3a95834f24424f39a63829, caeceed8f2d298b4a871ac2f97acc77c8c6d147a or even f7145a6f96a5f8a4322b1cf614396d3fd8ac07bf.

I propose some CI check for it, otherwise it will just happen again.

DagBruck, 2023-03-28:
> I propose some CI check for it, otherwise it will just happen again.

Probably a good idea, but note that "weird" numbers such as 1e-10 may occur naturally as parameters. At least the risk that Dymola adds the garbage has been reduced.

TManikantan, 2023-05-25:
Will close this issue, refer #4139 (used for missing commits) and #4107

closed 2023-05-25 by TManikantan